realthunder/FreeCAD_assembly3 #88: Sketcher: Failed to validate broken face and/or Crash

ceremcem, 2018-09-13:
# Description 

This problem occurs when we use external geometry via sub-shape binder and then solve the constraints.

# Reproduction 

1. Get a solid (`A`)
1. Draw a sketch (`Sketch`) using constraints to the external geometry (`ExtConstraints`) via sub-shape binder (`Binder`) of `A` 
2. Pad it to create a solid (`B`)
3. Create a constraint between `A` and `B`
4. Disable "Auto Recompute" (do not solve constraints automatically)
5. Move `A`
6. Click "Refresh" button (at the top)
6. See everything is normal and `B` is moved accordingly (Because when we move `A`, `Binder` is moved accordingly, thus the `ExtConstraints` are moved, thus `Sketch` is moved/re-drawn to/at the correct position. 
7. Press "Solve constraints"

Then one of the followings may happen: 
### You'll end up with broken face: 

[image]

**How to fix**: Open the sketch in edit mode, then close. You don't need to fix anything here. 

### or with a crash: 

```
(previous outputs seem irrelevant)
Link to Edge10 already exists in this sketch.
Failed to add external geometry: Not able to add external shape element
Link to Edge10 already exists in this sketch.
Failed to add external geometry: Not able to add external shape element
Selected external reference plane must be normal to sketch plane
Failed to add external geometry: Not able to add external shape element
Selected external reference plane must be normal to sketch plane
Failed to add external geometry: Not able to add external shape element
Link to Edge14 already exists in this sketch.
Failed to add external geometry: Not able to add external shape element
Link to Edge20 already exists in this sketch.
Failed to add external geometry: Not able to add external shape element
Updating geometry: Error build geometry(5): 
Invalid solution from DogLeg solver.
Drag curve: Not able to move point with the id and type: (3, 0)
Drag curve: Not able to move point with the id and type: (3, 0)
<asm3.sys> 0.000211 - solver.py(119): List of failed constraint:
"Constraint005(hat-offset)"<PlaneAlignment>, handle: 2
Updating geometry: Error build geometry(4): 
Invalid solution from DogLeg solver.
Updating geometry: Error build geometry(5): 
Invalid solution from DogLeg solver.
1.02e-07 Tree.cpp(2595): [ComboView] sub object not found Sketch009.;H_Axis.
Program received signal SIGSEGV, Segmentation fault.
```

**On re-start**, you'll see a recovery screen. Click "recover", wait for a while and see the documents are loaded in the tree. You'll end up with broken links. Close the project, say "yes" to "should we save the file?" prompts, then re-open the project. 

### Incorrect Placement (mostly this one happens)

You may end up with incorrect placement, because `Binder` isn't moved to the correct position [(screencast)](https://user-images.githubusercontent.com/6639874/45477972-8b29c200-b74b-11e8-9377-777397ef6091.gif).

**Fix**: Double click the `Binder` in the tree, it will be updated to the correct position, thus the sketch will be updated. Click "Reload" button at the top, and `B` is placed to the correct position.

realthunder, 2018-09-13:
Anything after the SIGSEGV? Those are the import ones.

ceremcem, 2018-09-13:
Nope. Nothing more at the bottom.

ceremcem, 2018-09-13:
I tried the same steps a few more times, even with the project snapshots taken before the crash (old version of project file) but there is no crash happening in those attempts.