realthunder/FreeCAD_assembly3 #89: How do we simulate double-click on sub-shape binder with Python API?

ceremcem, 2018-09-13:
I want to update all sub-shape binders with a macro. Apparently it's not `.recompute()` method. What method of a Sub-shape Binder updates it like we double-clicked on it?

ceremcem, 2018-09-26:
I needed to update many sub-shape binders manually, so here I'm again. The problem with above approach is that we need to know the source of a sub-shape manually, true? IMO, that makes writing a script to update every sub-shape binder impossible. Am I missing something here?

realthunder, 2018-09-27:
I just realized that `doubleClicked()` API is not exposed to Python. Anyway, it's been fixed.

Back to the topic. What you need is actually the top-leve parent groups that contain the binder and the object hierarchical path that leads to it. It is not an easy task. And there may be ambiguities that cannot be resolved using scripts. For example, imaging a binder inside an assembly, and there is a link to that assembly. Now the same binder appears in two different places. Which one to use for placement adjustment cannot be resolved by scripts automatically. Here is the script to find all top level groups and the corresponding object path, plus a function to simulate double clicking.

```python
from collections import defaultdict
def find_path(parent,tp,exclude):
    ret = []
    for sub in parent.getLinkedObject(True).getSubObjects(1):
        obj = parent.getSubObject(sub,retType=1)
        if not obj or obj in exclude:
            continue
        if obj.isDerivedFrom(tp):
            ret.append(sub)
        else:
            for ssub in find_path(obj,tp,exclude):
                ret.append(sub+ssub)
        exclude.add(obj)
    return ret

def find_objs(tp='PartDesign::SubShapeBinder'):
    objs = App.ActiveDocument.Objects
    ret = defaultdict(list)
    exclude = set()
    for obj in objs:
        # only checking group type objects
        if not obj.hasChildElement() and \
           not obj.hasExtension('App::GeoFeatureGroupExtension'):
            continue
        if obj in exclude:
            continue
        subs = find_path(obj,tp,exclude)
        if subs:
            ret[obj] += subs

    # filter out non-root-level groups
    for obj in exclude:
        ret.pop(obj,None)

    # find root-level links to those root-level groups. Omit this step if you
    # don't want to include root-level links
    filtered = [ o for o in objs if o not in exclude and o not in ret ]
    for obj in filtered:
        subs = ret.get(obj.getLinkedObject(True),None)
        if subs:
            ret[obj] = subs
    return ret

def update_objs(tp='PartDesign::SubShapeBinder'):
    for obj,subs in find_objs(tp).items():
        for sub in subs:
            sobj = obj.getSubObject(sub,retType=1)
            if not sobj:
                continue
            Gui.Selection.clearSelection()
            Gui.Selection.addSelection(obj,sub)
            sobj.ViewObject.doubleClicked()
```

closed 2020-06-18 by ceremcem