realthunder/FreeCAD_assembly3 #95: FreeCAD Crash

ceremcem, 2018-09-14:
# Last Traceback 

```
<App::Document> Document.cpp(2290): 'usb_a_type_double_thd' object 'Link001' xlink property 'LinkedObject' time stamp changed
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 356, in execute
    self.getElementSubname())
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 875, in getElementInfo
    objName(parent), subnameRef))
<type 'exceptions.RuntimeError'>: Invalid sub-object "Parts", Assembly002.Elements002.$Element060.
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1016, in execute
    info = self.getInfo(True)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1194, in getInfo
    self.getElementSubname(),shape=shape)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1078, in getElementSubname
    raise RuntimeError('Element link broken')
<type 'exceptions.RuntimeError'>: Element link broken
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1016, in execute
    info = self.getInfo(True)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1194, in getInfo
    self.getElementSubname(),shape=shape)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1078, in getElementSubname
    raise RuntimeError('Element link broken')
<type 'exceptions.RuntimeError'>: Element link broken
1.04e-07 Selection.cpp(1482): Sub-object usb_a_type_double_thd#Assembly.Constraints.Constraint012.ElementLink025. not found
0.000460524 Selection.cpp(1482): Sub-object usb_a_type_double_thd#Assembly.Constraints.Constraint012.ElementLink026. not found
0.00780505 Tree.cpp(2629): [ComboView] sub object not found Constraint012.ElementLink026.
0.322197 Selection.cpp(1482): Sub-object usb_a_type_double_thd#Assembly.Constraints.Constraint018.ElementLink037. not found
0.322605 Selection.cpp(1482): Sub-object usb_a_type_double_thd#Assembly.Constraints.Constraint018.ElementLink038. not found
0.330415 Tree.cpp(2629): [ComboView] sub object not found Constraint018.ElementLink038.
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 356, in execute
    self.getElementSubname())
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 875, in getElementInfo
    objName(parent), subnameRef))
<type 'exceptions.RuntimeError'>: Invalid sub-object "Parts", Assembly002.Elements002.$Element060.
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 97, in addSelection
    self.onChange()
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 86, in onChange
    if not self.timer.isSingleShot():
<type 'exceptions.TypeError'>: function takes at least 2 arguments (1 given)
```

# Last actions 

I was editing a Sketch under a pad. I created a new Spreadsheet, write something in there. Then returned to the Sketch, double clicked to edit it, then this crash has happened. 

Note that I copied this document from a different (more complex) document (with "save a copy"), then removed irrelevant parts to separate USB ports into another file. 

# Attachments 

[usb-a-type-double-thd.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2383187/usb-a-type-double-thd.fcstd.remove-extension.zip)

ceremcem, 2018-09-14:
No traceback, but again crashed while entering edit mode for a sketch (the first pocket).

ceremcem, 2018-09-14:
# Reproduction: 

1. Open the file: [usb-a-type-double-thd.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2383287/usb-a-type-double-thd.fcstd.remove-extension.zip)
2. Double click Pocket/Sketch001
3. [Crash happens without traceback](https://user-images.githubusercontent.com/6639874/45550811-b4724d00-b834-11e8-90eb-3971b5eadead.gif)

ceremcem, 2018-09-14:
This time I tried to edit Pocket001/Sketch002 and here is the traceback: 

```
Sketcher::SketchObject: Links go out of the allowed scope
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f228131f060]
#1  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d0284) [0x7f228a19f284]
#2  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d0355) [0x7f228a19f355]
#3  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d337b) [0x7f228a1a237b]
#4  0x7f228a1a23ca in Swig_python::createSWIGPointerObj_T(char const*, void*, _object**, int) from /opt/FreeCAD/lib/libFreeCADBase.so+0x4d
#5  0x7f228a13b2d4 in Base::InterpreterSingleton::createSWIGPointerObj(char const*, char const*, void*, int) from /opt/FreeCAD/lib/libFreeCADBase.so+0x15e
#6  0x7f228bda5207 in Gui::ViewProviderPy::getRootNode() const from /opt/FreeCAD/lib/libFreeCADGui.so+0x5d
#7  0x7f228bd9ef0d in Gui::ViewProviderPy::staticCallback_getRootNode(_object*, void*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x7d
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(_PyObject_GenericGetAttrWithDict+0xbb) [0x7f2289ade0eb]
#9  0x7f228a17615a in Base::PyObjectBase::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0x12e
#10  0x7f228a10c0fe in Base::BaseClassPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xc8
#11  0x7f228a16efa8 in Base::PersistencePy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xc8
#12  0x7f228ac22022 in App::PropertyContainerPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0xc8
#13  0x7f228ab43aae in App::ExtensionContainerPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0xc8
#14  0x7f228bd9fb94 in Gui::ViewProviderPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xc8
#15  0x7f228bd8cf30 in Gui::ViewProviderDocumentObjectPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xc8
#16  0x7f2253de5194 in PartGui::ViewProviderPartExtPy::_getattr(char const*) from /opt/FreeCAD/lib/PartGui.so+0xc8
#17  0x7f22529cc7f4 in PartDesignGui::ViewProviderPy::_getattr(char const*) from /opt/FreeCAD/lib/PartDesignGui.so+0xc8
#18  0x7f228a175ce0 in Base::PyObjectBase::__getattro(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADBase.so+0x130
#19  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x488c) [0x7f2289abc3dc]
#20  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f2289abe390]
#21  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f2289abe390]
#22  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f2289abe390]
#23  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f2289c2729c]
#24  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f2289ab79c9]
#25  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyRun_StringFlags+0x76) [0x7f2289b6f196]
#26  0x7f228a138cfe in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xd2
#27  0x7f228ba3f8cc in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x9c
#28  0x7f228ba3f913 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, QByteArray const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x37
#29  0x7f225230b8e8 in SketcherGui::ViewProviderSketch::setEdit(int) from /opt/FreeCAD/lib/SketcherGui.so+0x69a
#30  0x7f228bd6cead in Gui::ViewProvider::startEditing(int) from /opt/FreeCAD/lib/libFreeCADGui.so+0x2d
#31  0x7f228bd8df13 in Gui::ViewProviderDragger::startEditing(int) from /opt/FreeCAD/lib/libFreeCADGui.so+0x2f
#32  0x7f228b9c2f36 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x133a
#33  0x7f22522f4de4 in SketcherGui::ViewProviderSketch::doubleClicked() from /opt/FreeCAD/lib/SketcherGui.so+0x38
#34  0x7f228bb8d60f in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x235
#35  0x7f228356fa00 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xe10
#36  0x7f2283933c8e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e
#37  0x7f2283a58ba3 in QAbstractItemView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e3
#38  0x7f2283a9a46c in QTreeView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3c
#39  0x7f2282a520a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#40  0x7f228351852c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#41  0x7f2283520ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#42  0x7f228ba0e9d7 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xff
#43  0x7f2282a51f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#44  0x7f228351eccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#45  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f228359a1a9]
#46  0x7f2283598b5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
[... cut ...]
#54  0x7f2282a50b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#55  0x7f2282a56bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#56  0x7f228b98ebfd in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x22cb
#57  freecad-git(main+0xa6f) [0x55602018c4f9]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f228130c2e1]
#59  freecad-git(_start+0x2a) [0x55602018b90a]
```

ceremcem, 2018-09-15:
I'm using Pivy from Debian repository and this is the current output: 
```console
$ sudo apt-get install python-pivy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-pivy is already the newest version (0.5.0~v609hg-3.1).
```

realthunder, 2018-09-15:
After you sync with my branch, do you still experience the same crash described above (editing Sketch of Pad)? If it is still repeatable, please compile FreeCAD with debug mode

```
cmake -DCMAKE_BUILD_TYPE=Debug -DFREECAD_USE_OCC_VARIANT="Official Version"
```

And try to run FC under gdb. This will give a more accurate stack trace on crash. After it crashes, type
`where` in gdb terminal to get the stack.

ceremcem, 2018-09-15:
In current build (`a179d7afebafb44771acc31045241ed7738cdd6f`) I can not reproduce the issue, so I'm going to close it. 

The debug tip is also great, I'm adding it to my builder scripts to get more accurate stack traces for further possible crashes.

ceremcem, 2018-09-15:
Sorry, trying to edit the Sketch under base Pad still produces the crash, I'm re-compiling FreeCAD in debug mode.

ceremcem, 2018-09-15:
```
(gdb) where
#0  0x00007fffee31cac7 in QTextEngine::itemize() const () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007fffee31ea6c in QTextEngine::attributes() const () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#2  0x00007fffee32b08f in QTextLine::layout_helper(int) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#3  0x00007fffee3ca050 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#4  0x00007fffee3cabb1 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#5  0x00007fffee3d037e in QCommonStyle::sizeFromContents(QStyle::ContentsType, QStyleOption const*, QSize const&, QWidget const*) const ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#6  0x00007fffee47d502 in QWindowsStyle::sizeFromContents(QStyle::ContentsType, QStyleOption const*, QSize const&, QWidget const*) const ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#7  0x00007fffee472b22 in QCleanlooksStyle::sizeFromContents(QStyle::ContentsType, QStyleOption const*, QSize const&, QWidget const*) const ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#8  0x00007fffee44c8cc in QGtkStyle::sizeFromContents(QStyle::ContentsType, QStyleOption const*, QSize const&, QWidget const*) const ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#9  0x00007fffee40aee9 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#10 0x00007fffee695204 in QStyledItemDelegate::sizeHint(QStyleOptionViewItem const&, QModelIndex const&) const () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#11 0x00007fffee60c425 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#12 0x00007fffee61825f in QListView::updateGeometries() () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#13 0x00007fffee10ce80 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#14 0x00007fffee4d1c8e in QFrame::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#15 0x00007fffee5f6ba3 in QAbstractItemView::viewportEvent(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#16 0x00007fffed5f00a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#17 0x00007fffee0b652c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#18 0x00007fffee0bd52c in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#19 0x00007ffff65984c1 in Gui::GUIApplication::notify (this=0x7fffffffd6c0, receiver=0x55555c213df0, event=0x7fffff803400)
    at /root/FreeCAD/src/Gui/GuiApplication.cpp:91
#20 0x00007fffed5eff1d in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#21 0x00007fffee1561be in QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#22 0x00007fffee105145 in QWidget::setGeometry(QRect const&) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#23 0x00007fffee555886 in QAbstractScrollAreaPrivate::layoutChildren() () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#24 0x00007fffee556dc8 in QAbstractScrollArea::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#25 0x00007fffee5f683b in QAbstractItemView::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#26 0x00007fffee0b654c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#27 0x00007fffee0bd52c in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#28 0x00007ffff65984c1 in Gui::GUIApplication::notify (this=0x7fffffffd6c0, receiver=0x55555c212f10, event=0x7fffff803bd0)
    at /root/FreeCAD/src/Gui/GuiApplication.cpp:91
#29 0x00007fffed5eff1d in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#30 0x00007fffee108d6a in QWidget::setContentsMargins(int, int, int, int) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#31 0x00007fffee555a40 in QAbstractScrollAreaPrivate::layoutChildren() () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#32 0x00007fffee61811c in QListView::updateGeometries() () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#33 0x00007fffee10ce80 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
---Type <return> to continue, or q <return> to quit---
 ```

realthunder, 2018-09-15:
`Return` to continue util no such prompt

ceremcem, 2018-09-15:
I was doing so but it's huge, I'll keep pressing.

realthunder, 2018-09-15:
The problem is still triggered by pivy, although indirectly. Please attach your swig generate header file. It is in your build directory, `src/Base/swigpyrun.h`.

ceremcem, 2018-09-15:
Is there any way to apply the `where` command after the prompt automatically so I could redirect the output to a file: 

```
gdb -ex=run --args freecad-git > log
```

realthunder, 2018-09-15:
There is no easy way, I think. Maybe [here](https://blog.cryptomilk.org/2010/12/23/gdb-backtrace-to-file/)?

ceremcem, 2018-09-15:
There are 5 files that are similar to `swigpyrun.h`: [swigpyrun_xxx.h.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2385566/swigpyrun_xxx.h.zip)

realthunder, 2018-09-15:
Also the coin.py file. Start freecad, type in the python console
```
from pivy import coin
coin.__file__
```
The coin.py is located at the same directory

realthunder, 2018-09-15:
The swigrun.h (named exactly as such) is located at your build directory, not the source directory. Say if you build directory is named as `build`, and it should be at `build/src/Base/swigrun.h`

realthunder, 2018-09-15:
oops, sorry, it is swigpyrun.h

ceremcem, 2018-09-15:
Aha! build directory! Okay, I'll add that. 

I started FreeCAD with `bt freecad-git` (by using above `.bashrc` alias) and it's been around 2-3 minutes after the crash, I think the dump process continues. I just interrupted the process, here is only the current output (3.9MB): https://gist.github.com/932adf39670e75aa70e334f3a1d0e628

[swigpyrun.h.txt](https://github.com/realthunder/FreeCAD_assembly3/files/2385582/swigpyrun.h.txt)

(coin.py is on its way)

realthunder, 2018-09-15:
No need the stack trace now. That crash is due to an over long warning message, because there is a python code PrintWarning() to report view, and forget to end with new line. However, the warning is caused by pivy.

I suspect the cause is that your pivy library is not built with the libcoin you are currently using. But you didn't have problems before, did you do any system upgrade?

ceremcem, 2018-09-15:
> I suspect the cause is that your pivy library is not built with the libcoin you are currently using

If it's a requirement, I may build it by myself too. 

>  did you do any system upgrade

No. Moreover, I'm running FreeCAD on an LXC container with `X Forwarding` where the LXC container is freshly installed and contains only FreeCAD and its dependencies.

ceremcem, 2018-09-15:
```py
>>> from pivy import coin
>>> coin.__file__
'/usr/lib/python2.7/dist-packages/pivy/coin.pyc'
```

[coin.py.txt](https://github.com/realthunder/FreeCAD_assembly3/files/2385589/coin.py.txt)

realthunder, 2018-09-15:
Can't find any problem. Just a shot in the dark. You build system is currently installed with swig version 3.0.10, while your pivy is built using 3.0.7. Can you uninstall swig and replace it with an older version? Then do a fresh build of FreeCAD, starting from cmake.

My coin.py is exactly the same as yours. However, my system swig is 3.0.8. So not sure if this is really the problem.

realthunder, 2018-09-15:
A second option is to manually build pivy from source.

ceremcem, 2018-09-15:
There is something weird. I uninstalled swig and then manually compiled with `configure && make && make install`: 

```console
# swig -version

SWIG Version 3.0.8

Compiled with g++ [x86_64-unknown-linux-gnu]

Configured options: +pcre
```

But Asm3 build proces finds swig version 3.0.10: 

```console 
# ./build-freecad-asm3/install-asm3.sh 
Clone (or update) Assembly3 Workbench
-------------------------------------
Cloning into '/opt/FreeCAD/Ext/freecad/asm3'...
remote: Counting objects: 1068, done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 1068 (delta 28), reused 40 (delta 15), pack-reused 1013
Receiving objects: 100% (1068/1068), 518.32 KiB | 542.00 KiB/s, done.
Resolving deltas: 100% (728/728), done.
Building SolveSpace
-------------------
Submodule 'slvs' (https://github.com/realthunder/solvespace.git) registered for path 'slvs'
Cloning into '/opt/FreeCAD/Ext/freecad/asm3/slvs'...
Submodule path 'slvs': checked out '8b66606ef0d30e3dc9038fa79bc8f1d46f96fed5'
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'zlib'
--   Found zlib, version 1.2.8
-- Checking for module 'libpng'
--   Found libpng, version 1.6.28
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.11.0
-- Checking for module 'json-c'
--   Found json-c, version 0.12.1
-- Checking for module 'freetype2'
--   Found freetype2, version 18.3.12
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.10") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
-- python version 2.7
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.13") 
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/FreeCAD/Ext/freecad/asm3/slvs/build
Scanning dependencies of target slvs_static_excp
[  8%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/util.cpp.o
[ 16%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/entity.cpp.o
[ 25%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/expr.cpp.o
[ 33%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/constraint.cpp.o
[ 41%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/constrainteq.cpp.o
[ 50%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/system.cpp.o
[ 58%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/unix/unixutil.cpp.o
[ 66%] Building CXX object src/CMakeFiles/slvs_static_excp.dir/lib.cpp.o
[ 75%] Linking CXX static library libslvs_excp.a
[ 75%] Built target slvs_static_excp
[ 83%] Swig source
Scanning dependencies of target _slvs
[ 91%] Building CXX object src/swig/python/CMakeFiles/_slvs.dir/slvsPYTHON_wrap.cxx.o
[100%] Linking CXX shared module _slvs.so
[100%] Built target _slvs
All done.
```

ceremcem, 2018-09-15:
Okay, it needed `apt autoremove` to remove `/usr/bin/swig3.0`.

ceremcem, 2018-09-15:
Okay, I only purged `/opt/FreeCAD/Ext/freecad/asm3` and compiled Asm3 WB from scratch against `swig 3.0.8` and the problem seems to be vanished.

realthunder, 2018-09-16:
I cannot reproduce the crash. Did you have the stack trace? Or, maybe elaborate on the steps.

realthunder, 2018-09-16:
Oh no, this is exact same crash as before, I mean the warning message too long bug, which in term is caused by pivy. It is back again? Please sync my branch, the warning too long should be fixed. Try it again, and please post the warning message.

ceremcem, 2018-09-16:
There seems no commit after my above comment. My current build is: 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15521 (Git)
Build type: Debug
Branch: LinkStage3
Hash: 4699af3da94bd5fa5dc3e551e5f956ecf0ee0f8c
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2018-09-16:
Any terminal output message?

ceremcem, 2018-09-16:
Here is the console output of last runs: 

```
FreeCAD 0.18, Libs: 0.18R15518 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
X Error: BadWindow (invalid Window parameter) 3
  Major opcode: 31 (X_GrabKeyboard)
  Resource id:  0x0
<TopoShape> TopoShapeEx.cpp(1928): Not all input shapes are mappable
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex4
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge6
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge8
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge7
<TopoShape> TopoShapeEx.cpp(1928): Not all input shapes are mappable
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex4
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge6
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge8
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge7
<TopoShape> TopoShapeEx.cpp(1928): Not all input shapes are mappable
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex4
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge6
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge6
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge9
<TopoShape> TopoShapeEx.cpp(1928): Not all input shapes are mappable
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex2
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex3
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Vertex4
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge1
<TopoShape> TopoShapeEx.cpp(2244): unnamed lower element Edge6
[... 288 lines cut ...]
#38  0x7f1bcfe09ba3 in QAbstractItemView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e3
#39  0x7f1bcfe4b46c in QTreeView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3c
#40  0x7f1bcee030a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#41  0x7f1bcf8c952c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#42  0x7f1bcf8d1ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#43  0x7f1bd7dab4c1 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xff
#44  0x7f1bcee02f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#45  0x7f1bcf8cfccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#46  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f1bcf94b1a9]
#47  0x7f1bcf949b5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#48  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7f1bcf973502]
#49  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f1bc63987f7]
[... cut ...]
#55  0x7f1bcee01b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#56  0x7f1bcee07bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#57  0x7f1bd7d2cbe2 in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x22fa
#58  freecad-git(main+0xa6f) [0x55cec9843559]
#59  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f1bcd6bd2e1]
#60  freecad-git(_start+0x2a) [0x55cec984296a]
Segmentation fault
```

realthunder, 2018-09-16:
It seems the crash reason changes once you run under gdb? Strange.

Anyway, when sees `Swig_python::createSWIGPointerObj_T` inside first few frames of the stack trace, it means it is caused by pivy.

And the warning message (which seems to only appear if you run under gdb), `Show.TempoVis.isIn3DObject error: Cannot find type information for requested type`, means swig cannot find some coin type, which should happen at all.

realthunder, 2018-09-16:
Can you double check your swigpyrun.h file. Open and see its version.

ceremcem, 2018-09-16:
Sure: 

```console
grep  Version ~/fc-build/src/Base/swigpyrun.h
 * Version 3.0.8
```

realthunder, 2018-09-16:
Please open freecad, new document, go to Part workbench, and create a box. Then type in python console

```
App.ActiveDocument.Box.ViewObject.SwitchNode
```

Any error message or crash?

ceremcem, 2018-09-16:
No crash, only an error message: 

```console
Traceback (most recent call last):
  File "<input>", line 1, in <module>
RuntimeError: Cannot find type information for requested type
```

ceremcem, 2018-09-16:
Upps! It was crashed! (only not immediately)

```
cat gdb.bt 
backtrace:
No stack.

registers:
The program has no registers now.

current instructions:
No registers.

threads backtrace:
```

I'll do it again.

ceremcem, 2018-09-16:
```console
root@fc2:~/fc-build# ldd lib/libFreeCADGui.so | grep Coin
	libCoin.so.80 => /usr/lib/x86_64-linux-gnu/libCoin.so.80 (0x00007ff10c2b4000)
root@fc2:~/fc-build# cd /usr/lib/python2.7/dist-packages/pivy/
root@fc2:/usr/lib/python2.7/dist-packages/pivy# ldd _coin.x86_64-linux-gnu.so | grep Coin
	libCoin.so.80 => /usr/lib/x86_64-linux-gnu/libCoin.so.80 (0x00007f0de0b47000)
```

realthunder, 2018-09-16:
No problem here.

realthunder, 2018-09-16:
FREECAD_USE_EXTERNAL_PIVY parameter not longer exists now, because every platform uses external pivy. By right your system pivy should just work. Don't know what's wrong here. Normally using python package should be straightforward. However, FreeCAD needs to pass its own native C++ object (Coin3D object) to pivy, which requires binary compatibility, hence the requirement of compatible SWIG version. Now, with your system, there seems to be something wrong with that binary compatibility.

realthunder, 2018-09-17:
Please sync my branch and repeat the Box test. If still failed. Please run `App.dumpSWIG()` in python console and post the terminal output here.

realthunder, 2018-09-17:
Please sync my branch, and run `App.dumpSWIG()`. I think we are getting close.

ceremcem, 2018-09-17:
`App.dumpSWIG()`:
```
<swig> 0, _p_Slvs_Constraint, Slvs_Constraint *
<swig> 0, _p_Slvs_Entity, Slvs_Entity *
<swig> 0, _p_Slvs_Param, Slvs_Param *
<swig> 0, _p_System, System *
<swig> 0, _p_allocator_type, allocator_type *
<swig> 0, _p_char, char *
<swig> 0, _p_difference_type, difference_type *
<swig> 0, _p_double, double *
<swig> 0, _p_int, intptr_t *|int *|int_least32_t *|int_fast32_t *|int32_t *|int_fast16_t *
<swig> 0, _p_long_long, int_least64_t *|int_fast64_t *|int64_t *|long long *|intmax_t *
<swig> 0, _p_p_PyObject, PyObject **
<swig> 0, _p_short, short *|int_least16_t *|int16_t *
<swig> 0, _p_signed_char, signed char *|int_least8_t *|int_fast8_t *|int8_t *
<swig> 0, _p_size_type, size_type *
<swig> 0, _p_std__allocatorT_unsigned_int_t, std::vector< unsigned int >::allocator_type *|std::allocator< unsigned int > *
<swig> 0, _p_std__invalid_argument, std::invalid_argument *
<swig> 0, _p_std__vectorT_unsigned_int_std__allocatorT_unsigned_int_t_t, std::vector< unsigned int,std::allocator< unsigned int > > *|std::vector< unsigned int > *|std::vector< Slvs_hConstraint > *
<swig> 0, _p_swig__SwigPyIterator, swig::SwigPyIterator *
<swig> 0, _p_unsigned_char, unsigned char *|uint_least8_t *|uint_fast8_t *|uint8_t *
<swig> 0, _p_unsigned_int, uintptr_t *|uint_least32_t *|uint_fast32_t *|uint32_t *|unsigned int *|Slvs_hEntity *|Slvs_hConstraint *|Slvs_hGroup *|uint_fast16_t *|Slvs_hParam *
<swig> 0, _p_unsigned_long_long, uint_least64_t *|uint_fast64_t *|uint64_t *|unsigned long long *|uintmax_t *
<swig> 0, _p_unsigned_short, unsigned short *|uint_least16_t *|uint16_t *
<swig> 0, _p_value_type, value_type *
```

realthunder, 2018-09-17:
That's it? Strange? Type `from pivy import coin`, and then `App.dumpSWIG()`

ceremcem, 2018-09-17:
Oh my! After importing `coin`, output is huge: https://gist.github.com/ceremcem/62f1459af15e916458e6d3bab7f14194

realthunder, 2018-09-17:
Then, repeat the box test.

ceremcem, 2018-09-17:
I can't understand. I had already repeated the test after syncing, and it was failing, so I didn't mention that at all.

Now I tried again, here is the output: 

```
>>> App.ActiveDocument.Box.ViewObject.SwitchNode
<pivy.coin.SoSwitch; proxy of <Swig Object of type 'SoSwitch *' at 0x7fff96874810> >
```

ceremcem, 2018-09-17:
Here is the same test, with and without importing `coin`: 

```
>>> App.ActiveDocument.Box.ViewObject.SwitchNode
Traceback (most recent call last):
  File "<input>", line 1, in <module>
RuntimeError: Cannot find type information for requested type
>>> from pivy import coin
>>> App.ActiveDocument.Box.ViewObject.SwitchNode
<pivy.coin.SoSwitch; proxy of <Swig Object of type 'SoSwitch *' at 0x7fffc2833990> >
```

realthunder, 2018-09-17:
So that the problem then! The auto module import is somehow not working. I'll take a closer look.

ceremcem, 2018-09-17:
Nice! It seems so.

ceremcem, 2018-09-17:
Importing coin does not seem to help at this point. Still crashing [while opening the sketch to edit](https://user-images.githubusercontent.com/6639874/45620606-27263700-ba86-11e8-9f5a-cc752de1ff40.png). Backtrace (partly, as it's huge, I interrupted the dump process): https://gist.github.com/9a95a26dff8ca749f6cb11b4aaef2ae1

realthunder, 2018-09-17:
Don't run it under gdb. See what happen.

ceremcem, 2018-09-17:
Same crash happens: 

```
aea@fc2:~$ freecad-git 
FreeCAD 0.18, Libs: 0.18R15533 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
X Error: BadWindow (invalid Window parameter) 3
  Major opcode: 31 (X_GrabKeyboard)
  Resource id:  0x0
Segmentation fault
aea@fc2:~$ 
```

realthunder, 2018-09-17:
You stack trace [here](https://gist.github.com/ceremcem/32203ff07b2f59b483763cf88b8e91ce) means that something is wrong with pivy.coin. I suggest you compile pivy yourself. Maybe try the fork [here](https://github.com/FreeCAD/pivy). You can use python distutils to build it. Make sure you build with debug symbol

```
python setup.sh build -g
```

ceremcem, 2018-09-17:
Would you help me building Pivy? Because I tried it (from the same repo) before and it threw (still throws) the error: 
```
 checking for SOQT via cmake
Traceback (most recent call last):
  File "setup.py", line 666, in <module>
    platforms=['Any']
  File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 588, in run
    self.pivy_configure()
  File "setup.py", line 453, in pivy_configure
    self.check_with_cmake()
  File "setup.py", line 229, in check_with_cmake
    if config_dict['SOQT_FOUND'] == 'false':
KeyError: 'SOQT_FOUND'
```

Which I thought I could install it by package manager: 
```
# apt-cache search soqt
libsoqt-dev-common - Qt GUI component toolkit for Inventor - common development files
libsoqt4-20 - Qt4 GUI component toolkit for Inventor - runtime
libsoqt4-dev - Qt4 GUI component toolkit for Inventor - development
```
I don't understand why those don't satisfy pivy's requirements but I'm having problems at this point.

realthunder, 2018-09-17:
According to [this](https://github.com/FreeCAD/pivy/issues/28#issuecomment-377968108), you either build libcoin yourself use cmake and install it, which will give you cmake-config file. Or, build pivy git tag 0.6.2 as mentioned in the comment.

ceremcem, 2018-09-17:
I successfully compiled and installed 0.6.2. Now I'm building against my pivy install.

ceremcem, 2018-09-17:
Crash still happens with or without `from pivy import coin` trick.

realthunder, 2018-09-17:
Can I have the stack track? Since now pivy.coin has debug information, we should be able to see more. Also, please confirm that FreeCAD is using your compiled pivy. 

```
from pivy import coin
coin.__file__
```

ceremcem, 2018-09-17:
This is the current [`/usr/local/lib/python2.7/dist-packages/pivy/coin.py`](https://gist.github.com/04f19e862230c342c2622bb1f5f6f04b).

1. Opening FreeCAD
2. `from pivy import coin`
3. Open the usb file 
4. Edit the first sketch in the first pad 
5. Crash, stack trace (interrupted):  https://gist.github.com/d781b3ca7d4be1ae5ce82473561c9ba5

[25 more replies]

closed 2020-12-02 by ceremcem