realthunder/FreeCAD_assembly3 #98: Spreadsheet still touched after recompute

ceremcem, 2018-09-15:
# Description 

While following https://github.com/realthunder/FreeCAD_assembly3/issues/52#issuecomment-421528392 I started to get the following exception (a few refreshes took place): 

```console
<App::Document> Document.cpp(2854): Spreadsheet still touched after recompute
Exception (Sat Sep 15 03:47:39 2018): Document::getDependencyList: The graph must be a DAG.  
<App::Document> Document.cpp(2854): Element002 still touched after recompute
<App::Document> Document.cpp(2854): Element001 still touched after recompute
<App::Document> Document.cpp(2854): Bend002 still touched after recompute
<App::Document> Document.cpp(2854): Element still touched after recompute
<App::Document> Document.cpp(2854): Sketch003 still touched after recompute
<App::Document> Document.cpp(2854): Sketch001 still touched after recompute
<App::Document> Document.cpp(2854): Body still touched after recompute
<App::Document> Document.cpp(2854): Pocket still touched after recompute
<App::Document> Document.cpp(2854): Spreadsheet still touched after recompute
<App::Document> Document.cpp(2854): Element003 still touched after recompute
```

Only difference between working version and erroneous version is placing the following line into `D6`: 

```
=Elements.<<$_bend_start.>>.Shape.Vertex1.Point.distanceToPlane(Elements.<<$front-face.>>.Shape.BoundBox.Center,Elements.<<$front-face.>>.Shape.Face1.Surface.Axis)
```

Calculation is correct, but because of above exception, recompute (thus the workflow) is broken.

# Attachments 

**Working**: [usb-20180915T1045.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2385495/usb-20180915T1045.zip)

**Erroneous**: [usb-20180915T1048.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2385496/usb-20180915T1048.zip)

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15518 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: a179d7afebafb44771acc31045241ed7738cdd6f
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2018-09-15:
You have introduced a dependency loop with that expression. That spreadsheet contains all primary design constant that drives various body feature of yours, while the distance expression can only be evaluated after it's referenced features has been recomputed, hence the loop as shown below

```
Spreadsheet recompute -> feature recompute -> Element recompute -> The distance expression cell touched -> Spreadsheet recompute
```

The solution is to put that expression in a separate sheet.

ceremcem, 2018-09-15:
This still creates a cyclic dependency: 

[image]

Can you provide a fixed version of "not working" version as an example?

realthunder, 2018-09-16:
Can you please send me your file of that screen capture? And also please clarify your intention of the calculated distance to plane. Where will it be used?

ceremcem, 2018-09-16:
The distance I want to dynamically measure is: 

[image]

I'm using it for calculating rest of the body length (the yellow part in the above screenshot) so that I can set the total depth of the socket to a fixed value no matter how I change the material thickness or bend radius.

# Attachments 

[usb-a-type-stacked-thd__testing__.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2386314/usb-a-type-stacked-thd__testing__.fcstd.remove-extension.zip)

realthunder, 2018-09-16:
This is not possible. You want to set the length of an early feature based on later features. That is clearly a dependency loop.

In you current use case, you should not use `Element` at all. Since all you referenced geometry elements are within the same body, you can directly reference those features. The usage of `Element` here is meant be similar to binder, that is, to bring in features across external coordinate space. For example, the front face element links to the `Parts` directly, and indirectly through a subname reference `Body.Pad001.Face52`. This means, anything inside `Parts` **must not** referencing the `Element` object, or else there will be loop. In short, child object cannot reference parent.

Solutions I can think of

* Either you make radius and thickness as prime design parameter, and directly calculate the distance based on these parameters.

* Or, you create an extra pad as a feature after the last bend, only then you are allowed to use the distance, as it is calculated from early features. But first, you need to rewrite distance expression by referencing features in body directly, or if you want similar benefit of an abstraction, you can use binder.

ceremcem, 2018-09-20:
> In you current use case, you should not use Element at all

I think we should always use the elements for this (dynamic measurement) purpose (even it's measured within the same object) because the object can be replaced by another object **or even** another object that is an assembly container so the related edges/faces/vertices would now belong to two distinct objects under the hood. I think we should be prepared for this case.

(to be continued)

sedatkrn, 2021-12-28:
i have smillar error for spreadsheet  
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Turkish/Turkey (tr_TR)

İ want to make a parametric U  shaped stair. it is runing but suddenly began to gave Error .

[image]

Video:   [https://youtu.be/MG5or3cSk7A](url)

file:  [https://drive.google.com/file/d/18mXbSt3jEzu_qCGSx49QLKPkUefx3qKm/view?usp=sharing](url)