realthunder/FreeCAD_assembly3 #101: Sync view disable key

ceremcem, 2018-09-16:
Sync view is a nice feature, but creates conflicts for some tools that require to select some target object while its own object is opened and then pressing one of its command buttons (like TechDraw Projections). 

It would be nice to have a key (like `Alt` or `Ctrl`) to disable this `Sync view` feature temporarily (while holding that key down).

realthunder, 2018-09-17:
I have exposed the sync view option as a toolbar button, with default accelerator keys as "V, C". This means that you can easily toggle this option by simply type V+C in your keyboard.

closed 2018-09-17 by realthunder

ceremcem, 2018-09-19:
[image]