realthunder/FreeCAD_assembly3 #102: Redundant constraints may lead incorrect solutions

ceremcem, 2018-09-16:
Reproduction and an example will be added soon. 

There is a situation I guess caused by indirectly redundant constraints. Currently not harmful, I'm handling them.

1. Create a part (A)
2. Create part with part design (B) whose sketch is drawn by external geometry constraints (EX) which belongs to the Binder (BN) which is a sub-shape binder of A
3. Create a constraint (PlaneAlignment, for example) between top surface of B and top surface of A, which the EX constraint is also in this plane (indirectly)
4. After things are moved and BN needs to be updated by double-clicking on it, there starts a race condition. 

At step 4, the B's top surface is in a different location than top surface of A, *I think* because:

* Asm3 wants to make them equal with top surface of A
* PartDesign external geometry redraws the sketch according to BN, which is in a different location. 

## Current Workaround 

Delete/disable Asm3 constraint and let the PartDesign's EX handle the relocation.

realthunder, 2018-09-17:
I don't think that is a problem of the program. This is the binder update cycle I talked about. For a correctly constrained system, the binder will eventually stop moving. What's your expected behavior?

ceremcem, 2018-09-17:
>  the binder will eventually stop moving

Will that happen when "the" calculated difference is below the float precision? If that's the case, this is something that impacts the performance if sub-shape binders are heavily used (like in my case) since it has *a lot of* re-computations. 

> What's your expected behavior

I'm thinking on this now.

ceremcem, 2018-09-18:
**IMPORTANT** Note to myself (and further users): Removing Asm3 constraints might (will) lead to incorrect solutions which will ruin your design (I remember @realthunder said something like that before, but I can't remember on which topic that was)

realthunder, 2018-09-18:
Really? I didn't remember either? Ruin your design, how? I am sorry, I am busy investigate some other bug, so didn't follow the binder issue, yet.

ceremcem, 2018-09-18:
"Ruining" will depend if you spot the issue or not. If you can't realize that some parts are moved without your intention, it will lead a ruined model. In my case, a part is moved too much so that I could spot the change immediately and enabled those asm3 constraints, now everything is back to normal.

realthunder, 2018-09-18:
Ah, I see. Yes, as long as there is DOF left, it is possible and in fact legal for the solver to move your parts. Even if there is no DOF left, there may be multiple solutions, such as different orientation of the part that all fit the constraints. You may want to freeze an assembly whenever possible.

ceremcem, 2018-09-27:
Another note to further readers (and myself): If sub-shape binder jumps whatever you do (i.e. update it with its source, recreate, etc.) check the race condition between external geometry and asm3 constraints. 

If external geometry wants to draw an object in a position and asm3 (solvespace) wants to move the part, asm3 moves the part **and** the subshape binders. 

In other words, when sub-shape binder is used for external geometry constraints source and resulting object's placement is changed by asm3 constraints, this possibly creates a race condition. 

# Example 

In the picture below, the Body001 is based on the Binder which binds to Body. There is a conflict created on purpose: The Body001's upper face touches the binder and lower face is set to PlaneAlignment constraint which wants to put the Body001.lower_face to Body.lower_face which is also Body001.upper_face, which absolutely creates the conflict: 

[image]

ceremcem, 2021-02-14:
Closing as this is an old issue and things have been changed dramatically in the newer version, which makes the issue impossible to reproduce.

closed 2021-02-14 by ceremcem