realthunder/FreeCAD_assembly3 #103: Moving objects another way from drag and drop

ceremcem, 2018-09-16:
In large trees, dragging&dropping objects (`Elemet`s, regular objects, `Element`s to create links in another document) becomes frustrating. 

### Proposal

Support Cut+Paste and Copy+Paste for 
* Move objects into Assembly containers 
* Create element links between documents

realthunder, 2018-09-23:
Select the object you want to drag as before, and navigate to the item you want to drop, then use key shortcut `Ctrl+Shift+D` to activate dragging.

closed 2018-09-23 by realthunder

realthunder, 2018-09-25:
I have changed the shortcut of dragging to `T, D`

ceremcem, 2018-09-25:
Sometime later we may do something about shortcuts, though. Different layouts makes the shortcuts longcuts.

realthunder, 2018-09-26:
I have exposed all option commands to command manager. You can customize command shortcut by right click in toolbar, and select `Customize`. Go to `Keyboard`. The commands are under category `TreeView`.