realthunder/FreeCAD_assembly3 #108: AttributeError: 'module' object has no attribute 'isCommandActive'

ceremcem, 2018-09-16:
The error in #105 is quite reproducable: 

1. Open FreeCAD 
2. Right click in the comboview 
3. Following exception is thrown: 

```
<asm3.main> 17.805456 - init_gui.py(71): 
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/FCADLogger.py", line 83, in _catch
    return func(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/init_gui.py", line 64, in _contextMenu
    name = cmd.getContextMenuName()
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 210, in getContextMenuName
    if cls.IsActive() and cls._contextMenuName:
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 682, in IsActive
    return FreeCADGui.isCommandActive('Std_LinkSelectLinked')
AttributeError: 'module' object has no attribute 'isCommandActive'

```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15521 (Git)
Build type: Debug
Branch: LinkStage3
Hash: 4699af3da94bd5fa5dc3e551e5f956ecf0ee0f8c
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2018-09-17:
Sync my branch

ceremcem, 2018-09-17:
This error has gone after syncing.

closed 2018-09-17 by ceremcem