realthunder/FreeCAD_assembly3 #109: TopoShapeEx.cpp: hasher mismatch

ceremcem, 2018-09-16:
# Reproduction 

1. Copy a part
2. Paste into the same assembly container 
3. Part is created in the root. 
4. Drag and drop the part assembly container again
5. The exception is displayed in the Report view: 

         <TopoShape> TopoShapeEx.cpp(413): hasher mismatch

6. Refresh 
7. Gives the exception in #105 

    ```
    <unknown exception traceback><type 'exceptions.SyntaxError'>: ('invalid syntax', ('<string>', 1, 21, '_path_value_temp_ = \n'))
    ```

luzpaz, 2021-02-20:
@ceremcem is this still a problem. I tried to recreate it but I ran in to exceptions just trying to open files. 

OS: Manjaro Linux (KDE//usr/share/xsessions/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.0205.23662 +3201 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 9391fcd19568d635bb290db09f9d616e6acdc4f4
Python version: 3.8.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

ceremcem, 2021-02-20:
This issue can't be reproduced because we had a breaking change, like 8-10 months ago. That attachment can not be opened with recent versions of FreeCAD. I'm closing the issue since the reproduction is not possible.

closed 2021-02-20 by ceremcem