realthunder/FreeCAD_assembly3 #119: Wrong number or type of arguments for overloaded function

ceremcem, 2018-09-18:
[image]

This error is occurred when I tried to use `Perpendicular` constraint for the two screws: 

[image]

realthunder, 2018-09-18:
Can I have the complete error message in the ReportView?

ceremcem, 2018-09-18:
Sorry, I thought that was the full error. Here is Report view output: 

```
<asm3.gui> 701.89362 - gui.py(282): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/FCADLogger.py", line 106, in report
    return func(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 405, in solve
    return _solve(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 375, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 72, in __init__
    ret = Constraint.prepare(cstr,self)
  File "/opt/FreeCAD/Ext/freecad/asm3/constraint.py", line 488, in prepare
    return mcs.getProxy(obj).prepare(obj,solver)
  File "/opt/FreeCAD/Ext/freecad/asm3/constraint.py", line 1148, in prepare
    ret = system.addPerpendicular(e1,e2,group=solver.group)
  File "/opt/FreeCAD/Ext/freecad/asm3/py_slvs/slvs.py", line 870, in addPerpendicular
    return _slvs.System_addPerpendicular(self, l1, l2, wrkpln, group, h)
NotImplementedError: Wrong number or type of arguments for overloaded function 'System_addPerpendicular'.
  Possible C/C++ prototypes are:
    System::addPerpendicular(Slvs_hEntity,Slvs_hEntity,Slvs_hEntity,Slvs_hGroup,Slvs_hConstraint)
    System::addPerpendicular(Slvs_hEntity,Slvs_hEntity,Slvs_hEntity,Slvs_hGroup)
    System::addPerpendicular(Slvs_hEntity,Slvs_hEntity,Slvs_hEntity)
    System::addPerpendicular(Slvs_hEntity,Slvs_hEntity)
```

commit 170e7396 (2018-09-18): constraint: fix Perpendicular constraint

Fixes #119
closed 2018-09-18 by realthunder