realthunder/FreeCAD_assembly3 #123: asm3CmdAxialMove exception

ceremcem, 2018-09-18:
# Traceback
```
Running the Python command 'asm3CmdAxialMove' failed:
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 370, in Activated
    mover.movePart(False,AsmCmdMove._moveInfo)
  File "/opt/FreeCAD/Ext/freecad/asm3/mover.py", line 302, in movePart
    vobj.Proxy._movingPart = AsmMovingPart(moveInfo.Hierarchy,info)
  File "/opt/FreeCAD/Ext/freecad/asm3/mover.py", line 30, in __init__
    rot = utils.getElementRotation(shape)
  File "/opt/FreeCAD/Ext/freecad/asm3/utils.py", line 366, in getElementRotation
    return getEdgeRotation(edge)
  File "/opt/FreeCAD/Ext/freecad/asm3/utils.py", line 356, in getEdgeRotation
    if not axis:

local variable 'axis' referenced before assignment
```

realthunder, 2018-09-18:
This error and also #122 only happens if you are using some unusual shapes. Can I have the file for those shape you are using. The unusual shapes requires numerical approximation to obtain its center and orientation. If possible, you should always use simple shapes for constraining, like an sketch element or datum object, which is also more robust against lost of reference due to model change.

closed 2018-09-18 by realthunder

ceremcem, 2018-09-18:
File is a logo DXF file, which has to be handled as is: 
[scada-gateway.dxf.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2394878/scada-gateway.dxf.zip)

realthunder, 2018-09-18:
Well, as shown in its feature name, they are all bezier curve. It has to go through a series conversion/approximation in order to be useable. How do you plan to constrain it?

ceremcem, 2018-09-18:
I don't know. Can't I use it like any sketch? (this is a real newbie question :)

realthunder, 2018-09-19:
I mean how you want to constrain it using asm3? You can easily link group it with a sketch, and use external geometry inside sketch to simplify the shape. You don't have to replicate the whole logo, obviously. Just use some lines or circles as markers for asm3 to constrain, similar to my suggestion [here](53#issuecomment-419344910)

commit f8dfcd66 (2018-09-18): utils: fix getElementRotation

Fixes #123