realthunder/FreeCAD_assembly3 #125: Understanding DOF remaining report

ceremcem, 2018-09-19:
With https://github.com/realthunder/FreeCAD_assembly3/commit/af5789c4f86b9a6d0b9771fe4f70dfd46cd523f9 "dof remaining: xxx" messages can be viewed in Report view. What is that report meaning? I thought it says "your objects have that amount of DOF, so if it's greater than zero, you will be end up with a messed model eventually. You have been warned" but it seems that it's not the case. 

# Reproduction 

1. Add a new object to a fully constrained model. "dof remaining" is "0". Expected DOF is 6 (or maybe 0, since solver might thing it has nothing to do with this object)
2. Create a PlaneAlignment constraint. Expected DOF is 1 (6-5). Reported DOF is still zero. 

What is that output meaning?

realthunder, 2018-09-19:
> so if it's greater than zero, you will be end up with a messed model eventually. You have been warned" but it seems that it's not the case.

That's not the case indeed. Imaging you have screws in your assembly. Leaving them one rotational DOF won't mess up anything.

> 2. Create a PlaneAlignment constraint between the new object's face and a face in the model. Expected DOF is 1 (6-5). Reported DOF is still zero.

Can you show a file with this. Two box with a PlaneAlignment will report DOF 3, two positional and one rotational.

ceremcem, 2018-09-19:
> Leaving them one rotational DOF won't mess up anything. 

Yes, my mistake. 

> Can you show a file with this. 

[Previous file](https://github.com/realthunder/FreeCAD_assembly3/files/2394886/case-v3-20180919T0256.zip) will suffice. If you add a `dc-power-jack` into the `enclosure` document and create a PlaneAlignment constraint, you will notice the dof report: [screencast](https://user-images.githubusercontent.com/6639874/45756977-f5020a00-bc2a-11e8-82f9-7cd18f1e0314.gif)

ceremcem, 2018-09-19:
> Leaving them one rotational DOF won't mess up anything.

Idea: If we had a chance to define a "safe free axes", then DOF output can also ignore these safe free axes, so it could perfectl give us "your model might be ruined" tip.

ceremcem, 2021-08-24:
@realthunder This issue/feature has the same importance as the "Unconnected wire count" in a PCB design software. 

This is a critical feature because if we leave any free object within the model, it can be moved by the solver in later editions, which will create conflict with our already produced mechanical parts.