realthunder/FreeCAD_assembly3 #126: Understanding the "Relations" branch in the tree view

ceremcem, 2018-09-20:
I've started using the "Relations" branch in the tree view and I found it very handy. (at first, I was afraid to touch it because when you create an Assembly container, it doesn't exist in the first place)

[image]

I didn't find anything about it in the wiki (or I missed something), so I thought it would be a start to share my understanding. 

`Relations` are like `Views` in databases. When we click the "Show me the related constraints" button, all related constraints are shown as grouped: 

[image]

This is a very nice and handy facility to navigate in a large model and reduce the number of Elements, thus keep the model simpler. 

# Example use case

When you want to place a second bolt in the same axis of first one, but on the other side of the device, naturally rotated 180 degrees. 

### Problem

If you click the first bolt's head (you have to, because you are not able to see anything other than the head at the moment) and click any of the cylindrical face/edge, you can use AxialAlignment constraint. That will do the job, but you will end up with a second, unneeded `Element` created because you used an unnecessary face while clicking the head of first bolt. 

### Solution: 

Click the head of the first bolt you are seeing, click the "Show me the related constraints" button. You will see 1 or 2 constraints, depending on your assembly/manufacturing approach. Select an appropriate ElementLink, click the appropriate face/edge of the second bolt, click the AxialAlignment constraint. 

You have saved 1 element.

realthunder, 2018-09-20:
The relation group is introduced [here](../wiki/Navigation). You probably missed this one?

ceremcem, 2018-09-20:
> You probably missed this one 

Yes, see my edit.