realthunder/FreeCAD_assembly3 #127: How to quickly remove an object from the assembly?

ceremcem, 2018-09-20:
How can we quickly delete an object along with its Elements and the related constraints that uses those unnecessary elements from the assembly? My current approach is:

1. Delete the object 
2. Delete the elements those are indicating error 
3. Delete the constraints those are indicating error
4. Delete the links those are indicating error

realthunder, 2018-09-20:
Well, you just discovered relation group. Select the part object, and click that command (A, R) to go to its corresponding relation object. Delete the relation along with all the constraints under it. There is no auto delete of elements because of their special purpose. You can use other navigation buttons to explore the link and its linked objects.

ceremcem, 2018-09-20:
So the trick is deleting the constraints first.

closed 2018-09-20 by ceremcem

realthunder, 2018-09-20:
You can select the relation and the constrains within to delete them in one go.

ceremcem, 2018-09-20:
Thanks for the tip!