realthunder/FreeCAD_assembly3 #132: FreeCAD crash

ceremcem, 2018-09-21:
# Last actions 

I copied an assembly container and I was about to paste it into another assembly container. Mistakenly, instead of selecting "Paste", I selected "Delete"

# Traceback 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7fbbea9450c0]
#1  0x7fbbc53a2c90 in vtable for NCollection_BaseList from /opt/FreeCAD/lib/Part.so+0x10
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15546 (Git)
Build type: Release
Branch: LinkStage3
Hash: 169aeb74d470d376011ac2309546ec57813cdc16
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2018-09-21:
The crash happened in another thread, where OCCT is doing something. This most likely is a OCCT bug. Can you reproduce this?

luzpaz, 2021-02-20:
Does this issue still occur on OCC 7.5.1 ?

ceremcem, 2021-02-20:
There is no way to test it against the file I mentioned (or meant), because of the same reason as [this one](https://github.com/realthunder/FreeCAD_assembly3/issues/109#issuecomment-782694949).

closed 2021-02-20 by ceremcem