realthunder/FreeCAD_assembly3 #143: Design In Assembly - Sketcher - Switch between section and full view

Nemesis81, 2018-10-03:
when editing a sketch from the assembly file. the "Switch between section and full view" function doesn't work with this message in the repor view:
`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCAhgze9d/usr/lib/freecad-daily/Mod/Show/TempoVis.py", line 406, in sketchClipPlane
    self.clipPlane(self.allVisibleObjects(sketch), enable, sketch.getGlobalPlacement(), 0.02)
  File "/tmp/.mount_FreeCAhgze9d/usr/lib/freecad-daily/Mod/Show/TempoVis.py", line 365, in clipPlane
    raise ValueError("Document object to be modified does not belong to document TempoVis was made for.")
<type 'exceptions.ValueError'>: Document object to be modified does not belong to document TempoVis was made for.
`
https://youtu.be/rNo95XRllS4

ceremcem, 2018-10-03:
I would like to note that personally I couldn't understand the issue. Can you please 
1. Include steps to reproduce the problem 
2. Add your build version 
3. Attach the example file

realthunder, 2018-10-03:
Problem confirmed.

ceremcem, 2018-10-04:
Sorry, after watching in high resolution, steps were pretty obvious.