realthunder/FreeCAD_assembly3 #146: slvs import error causing "no solver backend found" on Linux

Kristaba, 2018-10-16:
I installed both FreeCAD with LinkStage3 and asm3 extension on Archlinux, based on available [AUR](https://aur.archlinux.org/packages/freecad-linkstage3-git/) with a few fixes.
However, at startup, a warning is displayed on the console:
```
<asm3.main> 0.271045 - init_gui.py(14): no solver backend found
```

After a bit of investigation, mainly through the Python console inside FreeCAD, it appear that I can't manually import `freecad.asm3.py_slvs.slvs` module:
```python
>>> import freecad.asm3.py_slvs.slvs
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/lib/freecad/Ext/freecad/asm3/py_slvs/slvs.py", line 17, in <module>
    _slvs = swig_import_helper()
  File "/usr/lib/freecad/Ext/freecad/asm3/py_slvs/slvs.py", line 16, in swig_import_helper
    return importlib.import_module('_slvs')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _slvs
```

As I have all the necessary files (`__init__.py`, `_slvs.so` and `slvs.py`) in the correct path (`/usr/lib/freecad/Ext/freecad/asm3/py_slvs`), I checked further the cause of this ImportError.
By looking on the auto-generated SWIG file, it appear to do the following:
- try to import a module named `freecad.asm3.py_slvs._slvs`
- if it fails with an ImportError, import `_slvs`

An attempt to load directly the native module fails with a different error:
```python
>>> import freecad.asm3.py_slvs.slvs
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/lib/freecad/Ext/freecad/asm3/py_slvs/slvs.py", line 17, in <module>
    _slvs = swig_import_helper()
  File "/usr/lib/freecad/Ext/freecad/asm3/py_slvs/slvs.py", line 16, in swig_import_helper
    return importlib.import_module('_slvs')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_slvs)
```

As I had no more ideas, I just followed the build instructions to replace my `py_slvs` content (so everything is freshly built, based on today's master branch), but it changed nothing.
I am not familiar with SWIG, but it seems that the error is caused to some miss-configuration somewhere (SWIG not generating the init_xxx function for some reason?).
Any idea of the root cause of the problem here?

ceremcem, 2018-10-16:
Maybe @salamandar should take a look at this. 

In the meantime, you may try to follow installation procedure semi-automatically: https://github.com/ceremcem/build-freecad-asm3/blob/master/install-asm3.sh

Salamandar, 2018-10-16:
Oh, it seems like python2 and python3 are mixed up in the dependencies. I had in mind that freecad supported py3.

realthunder, 2018-10-16:
Maybe you have both Python2 and 3 installed on your system. The CMake picks Python3. You can rerun cmake with additional switch `-DPYTHON_VERSION_STRING="2.7"` to specify the version you want.

Kristaba, 2018-10-17:
Thanks all, it was indeed an issue caused by `slvs` being compiled for Python 3 instead of Python 2.

@realthunder the `-DPYTHON_VERSION_STRING="2.7" did not fix it (it seemed to be simply ignored by `cmake`), I had to give the path to Python binary with `-DPYTHON_EXECUTABLE:FILEPATH='/usr/bin/python2'` instead.

@Salamandar this is a distro-specific discussion, but as we are all here it will be faster :)
I fixed this issue and made explicit all the dependencies required by `slvs` in the `assembly3` package directly. Here is the diff from your AUR PKGBUILD:
```diff
From 7eb2b79abc1be60119aefa0ea354dd0af6b616a3 Mon Sep 17 00:00:00 2001
From: Leo Grange <grangeleo@gmail.com>
Date: Wed, 17 Oct 2018 09:44:04 +0200
Subject: [PATCH] fix slvs for Python 2 and explicit dependencies

---
 PKGBUILD | 21 ++++++++++++++++++---
 1 file changed, 18 insertions(+), 3 deletions(-)

diff --git a/PKGBUILD b/PKGBUILD
index 7393442..2016f41 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -1,7 +1,7 @@
 # Maintainer: Salamandar <felix at piedallu dot me>
 
 pkgname='freecad-assembly3-git'
-pkgver=r247.f8dfcd6
+pkgver=r252.cb84709
 pkgrel=1
 pkgdesc="Assembly3 workbench for FreeCAD v0.18"
 arch=('x86_64')
@@ -15,7 +15,22 @@ sha256sums=(
 )
 depends=(
   'freecad-linkstage3-git'
-  'python'
+  'python2'
+  'libpng12'
+  'json-c'
+  'freetype2'
+  'fontconfig'
+  'gtkmm'
+  'pangomm'
+  'libgl'
+  'glu'
+  'glew'
+  'libspnav'
+)
+
+makedepends=(
+  'cmake'
+  'git'
 )
 
 pkgver() {
@@ -29,7 +44,7 @@ build() {
 
   mkdir slvs/build -p
   pushd slvs/build >/dev/null
-  cmake -DBUILD_PYTHON=1 -DOpenGL_GL_PREFERENCE=GLVND ..
+  cmake -DBUILD_PYTHON=1 -DPYTHON_EXECUTABLE:FILEPATH='/usr/bin/python2' -DOpenGL_GL_PREFERENCE=GLVND ..
   make _slvs
   cp src/swig/python/{slvs.py,_slvs.so} ../../py_slvs
   popd >/dev/null
-- 
2.19.0
```

closed 2018-10-17 by Kristaba