realthunder/FreeCAD_assembly3 #148: trouble compiling.

J-Dunn, 2018-11-28:
Hi , I'm having trouble compiling on Fedora29 with gcc 8.2.1 , I have FreeCAD-master  building fine so most deps should be satisfied. cMake runs OK but the build fails after about 30 min. :

```
[ 56%] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/View3DViewerPy.cpp.o
/svn/FreeCAD-LinkStage3/src/Gui/View3DInventorViewer.cpp: In member function ‘void Gui::View3DInventorViewer::viewSelection()’:
/svn/FreeCAD-LinkStage3/src/Gui/View3DInventorViewer.cpp:2825:49: error: ‘virtual void SoCamera::viewBoundingBox(const SbBox3f&, float, float)’ is protected within this context
         cam->viewBoundingBox(box,aspectratio,1.0);
                                                 ^
In file included from /usr/include/Coin3/Inventor/nodes/SoOrthographicCamera.h:28,
                 from /svn/FreeCAD-LinkStage3/src/Gui/View3DInventorViewer.cpp:56:
/usr/include/Coin3/Inventor/nodes/SoCamera.h:115:16: note: declared protected here
   virtual void viewBoundingBox(const SbBox3f & box, float aspect,
                ^~~~~~~~~~~~~~~
In file included from /usr/include/Coin3/Inventor/SbBox.h:33,

```

Any help on what I need to do to get this to build here? 

Many thanks.

J-Dunn, 2018-11-28:
Just to be clear, I'm having a problem building LinkStage3, not assembly3.

ceremcem, 2018-11-28:
What is the commit hash of LinkStage3 you are trying to compile?

> I have FreeCAD-master building fine so most deps should be satisfied

Nope, we can't say that. You may take a look at [scripts](https://github.com/ceremcem/build-freecad-asm3/tree/master/deps) which are used for installing dependencies.

realthunder, 2018-11-28:
The problem is cause by an out-dated libcoin installation on your system. Please check out the FreeCAD build instruction [here](https://www.freecadweb.org/wiki/CompileOnUnix), expand the Fedora section. Quoting here

> coin3-devel
> (if coin2 is the latest available for your version of Fedora, use packages from http://www.zultron.com/rpm-repo/)

You may also want to build pivy, the Python binding for coin
> pivy ( https://bugzilla.redhat.com/show_bug.cgi?id=458975 Pivy is not mandatory but needed for the Draft module)

J-Dunn, 2018-11-28:
thanks, you sound quite sure but what are you calling "out-dated". I do have Coin3:

```
# dnf install Coin3-devel Coin3
Last metadata expiration check: 0:05:40 ago on Wed 28 Nov 2018 23:36:44 GMT.
Package Coin3-devel-3.1.3-23.fc29.x86_64 is already installed.
Package Coin3-3.1.3-23.fc29.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!

# dnf remove Coin2-devel Coin2
No match for argument: Coin2-devel
No match for argument: Coin2
Error: No packages marked for removal
```

J-Dunn, 2018-11-28:
... plus I don't really get the point about pivy.  I have distro pivy . What is the relevance of the that old 2012 bug? 
```
dnf install python2-pivy
Last metadata expiration check: 0:22:00 ago on Wed 28 Nov 2018 23:36:44 GMT.
Package python2-pivy-0.5.0-21.hg609.fc29.x86_64 is already installed.
Dependencies resolved.
```
I built FreeCAD-master without specifying external_PIVY flags , so I was under the impression was rebuilt during that process. Is that different on your linkStage3 ?

closed 2018-11-29 by realthunder

realthunder, 2018-11-29:
I have just made a commit to fix build when using Coin3. You should be able to build now.

I also found a [thread](https://forum.freecadweb.org/viewtopic.php?style=4&f=4&t=9213) regarding the Coin version on Fedora. Apparently Fedora has no official build for Coin4. And since one of the member mentioned that there are known bugs in Coin3 that affects FC, you may want to build Coin4 yourself, from the official [repo](https://bitbucket.org/Coin3D/coin/wiki/Home). And if you choose to build Coin yourself, then you have to build pivy as well.

realthunder, 2018-11-29:
> I built FreeCAD-master without specifying external_PIVY flags , so I was under the impression was rebuilt during that process. Is that different on your linkStage3 ?

FreeCAD source no longer include pivy, so that flag should have no effect now. You can find pivy source [here](https://github.com/FreeCAD/pivy)

realthunder, 2018-11-29:
According to the error message, pivy setup is looking for `qmake`. I am not sure why it can't be found, as it should be installed along with qt dependency.

J-Dunn, 2018-11-29:
I hacked it to look for qmake-qt4 and it got further:

```

  File "setup.py", line 229, in check_with_cmake
    if config_dict['SOQT_FOUND'] == 'false':
KeyError: 'SOQT_FOUND'

```
I thought SoQt was not needed if I went down this way.  Do I need to install distro's SoQt again?

realthunder, 2018-11-29:
FC does not use SoQt, but official pivy use it. FC fork of pivy is supposedly patch to allow build without SoQt. But there seems to be some bug here in setup. Try change

```
if config_dict['SOQT_FOUND'] == 'false':
```

into

```
if config_dict.get('SOQT_FOUND','false') == 'false':
```

And you probably need to comment out the following two lines

```
        if not bool(self.cmake_config_dict['SOQT_FOUND']):
            raise(RuntimeError('soqt was not found, but you need soqt to build pivy'))
```

J-Dunn, 2018-11-29:
Am I the only person building and testing this ?? I guess that's a result of providing binaries. 

```
checking for COIN via cmake

checking for SOQT via cmake

disable soqt, because cmake couldn't find it
Traceback (most recent call last):
  File "setup.py", line 667, in <module>
    platforms=['Any']
  File "/usr/lib64/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib64/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 589, in run
    self.pivy_configure()
  File "setup.py", line 454, in pivy_configure
    self.check_with_cmake()
  File "setup.py", line 235, in check_with_cmake
    if not bool(self.cmake_config_dict['COIN_FOUND']):
KeyError: 'COIN_FOUND'

```
I built and installed Coin4 but  it also seems to get a null when looking for it. 
```
ls /usr/local/lib64/cmake/Coin-4.0.0
coin-config.cmake  coin-config-version.cmake  coin-export.cmake  coin-export-release.cmake

```

J-Dunn, 2018-11-29:
```
calling: cmake 
CMake Error at CMakeLists.txt:5 (find_package):
  Could not find a package configuration file provided by "Coin" with any of
  the following names:

    CoinConfig.cmake
    coin-config.cmake

  Add the installation prefix of "Coin" to CMAKE_PREFIX_PATH or set
  "Coin_DIR" to a directory containing one of the above files.  If "Coin"
  provides a separate development package or SDK, be sure it has been
  installed.

```
CMAKE_PREFIX_PATH was empty so I added the local Coin prefix:
```

ls  $CMAKE_PREFIX_PATH
cmake  libCoin.so  libCoin.so.4.0.0  libCoin.so.80  pkgconfig
```
still not finding coin4

J-Dunn, 2018-11-29:
OK, I think what you need to do here is ensure that list is not empty:

      `         for var in coin_vars + soqt_vars + [' ']:`
That's one set closer.
```

Checking for swig...
'/bin/swig'
Checking for SWIG version...
3.0.12
calling: cmake 

checking for COIN via cmake
COIN_FOUND: true
COIN_VERSION: 4.0.0
COIN_INCLUDE_DIR: /usr/local/include
COIN_LIB_DIR: /usr/local/lib64

checking for SOQT via cmake
SOQT_FOUND: false
SOQT_VERSION: 
SOQT_INCLUDE_DIR: 
SOQT_LIB_DIR: 

disable soqt, because cmake couldn't find it

coin-features are not supported in this version
Preparing Inventor headers:.

=== Generating pivy/coin_wrap.cpp for coin ===

swig -w302,306,307,312,314,325,361,362,467,389,503,509,510 -c++ -python -includeall -modern -D__PIVY__ -I. -Ifake_headers -I"/usr/local/include" -Iinterfaces  -o pivy/coin_wrap.cpp interfaces/coin.i
Inventor/SbDict.h:53: Error: Unable to find 'cstddef'
Inventor/SbString.h:40: Error: Unable to find 'cstdarg'
Inventor/lists/SbPList.h:41: Error: Unable to find 'cassert'
Inventor/lists/SbPList.h:42: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/lists/SbList.h:36: Error: Unable to find 'cassert'
/usr/local/include/Inventor/lists/SbList.h:37: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/elements/SoSubElement.h:51: Error: Unable to find 'cassert'
/usr/local/include/Inventor/misc/SoState.h:38: Error: Unable to find 'cassert'
/usr/local/include/Inventor/fields/SoSubField.h:39: Error: Unable to find 'cassert'
/usr/local/include/Inventor/SbBSPTree.h:36: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/elements/SoNormalElement.h:38: Error: Unable to find 'cassert'
/usr/local/include/Inventor/caches/SoCache.h:41: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/details/SoSubDetail.h:40: Error: Unable to find 'cassert'
/usr/local/include/Inventor/elements/SoProfileCoordinateElement.h:39: Error: Unable to find 'cassert'
/usr/local/include/Inventor/engines/SoSubEngine.h:36: Error: Unable to find 'cassert'
/usr/local/include/Inventor/misc/SoGLImage.h:47: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/nodes/SoExtSelection.h:39: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/SbClip.h:38: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/SbHeap.h:38: Error: Unable to find 'cstddef'
Inventor/SbImage.h:43: Error: Unable to find 'cstddef'
/usr/local/include/Inventor/SbTesselator.h:39: Error: Unable to find 'cstddef'
SWIG did not generate wrappers successfully! ** Aborting **

```

J-Dunn, 2018-11-29:
Thanks. Good questions. I had forgotten to remove Coin3 and Coin3-devel  distro pkgs. Having done that I redid the Coin4 make install just to be sure. ( You will note above that it was already installed ). 

This did not change the output and the above errors.

J-Dunn, 2018-11-29:
` `         for var in coin_vars + soqt_vars + [' ']:``
can you comment on that fix ? It prevents the null list causing errors. Here I have used and empty string instead of a space. This gives same output, where we see it does not error on SOQT and does detect COIN.  However, it goes on afterwards to say : 

> coin-features are not supported in this version

realthunder, 2018-11-29:
I think that should be fine. 

Those errors means the compiler couldn't find some standard c++ header file. That's strange. Maybe you can try the `setup.old.py`. This requires you to install SoQt. Just install your system supplied one. And try
```
python setup.old.py build
```

J-Dunn, 2018-11-29:
```

python setup_old.py build

                           ~~~ HISSSSSSSSSS ~~~
                           Welcome to Pivy 0.6.5a0!

Platform...linux2
Python version...2.7.15
Checking for swig...
'/bin/swig'
Checking for SWIG version...
3.0.12
Warning: Pivy has only been tested with the following SWIG versions: 1.3.31 1.3.33 1.3.35 1.3.40.
Checking for coin-config...
not found.

```
wouldn't it be best to find out why it is not finding the most important headers?

realthunder, 2018-11-29:
According to [this](https://github.com/FreeCAD/pivy/issues/42) issue, the missing header is caused by some recent Coin source changes. I guess the fake header is needed by Swig parser somehow. Maybe you can manually create empty files in the fake_headers directory.

J-Dunn, 2018-11-30:
I now have FreeCAD-master built and working. I would like to be able to test asm3 with coin4 since Fedora's packages seem badly out of date here.

ceremcem, 2018-11-30:
If you just want to try asm3 out (no special purposes on native
installation), I'd recommend the VM approach:
https://github.com/ceremcem/build-freecad-asm3

It's not only for testing purposes, but also has many invaluable advantages
for daily usage. (I'm using Asm3 regularly in this way, I've no other setup)

J-Dunn <notifications@github.com>, 30 Kas 2018 Cum, 12:21 tarihinde şunu
yazdı:

J-Dunn, 2018-11-30:
Thanks for pointing out the fake headers issue. I'll look into that. 

However, I think the  setup.py code needs some checking with error conditions. It seems code has been put in place to trap certain things being present but has only ever been run on a system where the errors are present. The code fails when they are not instead of reporting as intended.

```

checking for COIN via cmake

checking for SOQT via cmake
Traceback (most recent call last):
  File "setup.py", line 666, in <module>
    platforms=['Any']
  File "/usr/lib64/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/usr/lib64/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/usr/lib64/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "setup.py", line 588, in run
    self.pivy_configure()
  File "setup.py", line 453, in pivy_configure
    self.check_with_cmake()
  File "setup.py", line 229, in check_with_cmake
    if config_dict['SOQT_FOUND'] == 'false':
KeyError: 'SOQT_FOUND'

```

Also with the qmake search. When qmake is not found it crashes out with the null object error. Hacking for it to look for qmake-qt4 gets me moving but failure to find should not crash python.

J-Dunn, 2018-11-30:
`You mean you have built pivy already? And by FreeCAD-master, do you mean my LinkStage3 branch? To test asm3, you'll need to build solvespace python binding too. Checkout the instruction here.`

No, I did mean FreeCAD-master.  When I had problems here , I thought a good first step was to build master to ensure I had all relevant packages and I was not bugging you here for something that was not related to your changes. 

I have build and installed coin4 but still struggling with pivy. I think it's the headers issue but too many distractions here. I run through it again later.

J-Dunn, 2018-11-30:
Oh yes and this one , more untested code, it seems.  I commented out the test you said above but there is another test for 'SOQT_FOUND' which seems to produce a runtime error instead of reporting SoQt not found. 
```

  File "setup.py", line 453, in pivy_configure
    self.check_with_cmake()

  File "setup.py", line 229, in check_with_cmake
    if config_dict['SOQT_FOUND'] == 'false':
KeyError: 'SOQT_FOUND'

```

realthunder, 2018-11-30:
You should be able to compile my branch. Pivy is not need during compile time. It is needed in runtime by Draft workbench, and maybe others. 

And you can also report the pivy issue to its own repo as well.