realthunder/FreeCAD_assembly3 #149: Tutorial error: Document.cpp(331) cannot edit object Assembly not found in document

poleguy, 2018-12-03:
I tried to run the tutorial on Preview 0.8.6 on Windows, and when you try to move the cubes within the linked assembly they don't move but spit out an error:
"Document.cpp(331) cannot edit object Assembly... not found in document..."

https://github.com/realthunder/FreeCAD_assembly3/wiki/Common-Operations

How stale is the 0.8.6 windows binary? Might this be fixed in the source already? Does the tutorial work on the latest source?

ceremcem, 2018-12-04:
> Does the tutorial work on the latest source?

I can confirm that those tutorials (and more) are working on the latest source on Linux. Moreover, there are lots of changes since 0.8.6 in the source. @realthunder is preparing a new release nowadays. You can wait for that or compile from source (which I think compiling from source has lots of advantages).

...or, @realthunder will comment this to point out the exact issue.

realthunder, 2018-12-04:
This is indeed a bug. Thanks for reporting. It only happens if owner assembly of the moving part is external linked. In default cases, the part mover moves part in context of the top level assembly, which is usually not external linked. I guess that's why the problem is hidden for so long. I'll fix it soon.

commit 2f54b912 (2018-12-04): mover: fix moving part of external linked assembly

Fixes #149
closed 2018-12-04 by realthunder