realthunder/FreeCAD_assembly3 #150: PointsOnCircle doesn't accept multiple points

ceremcem, 2018-12-04:
While evaluating [the issue on freecad forum](https://forum.freecadweb.org/viewtopic.php?f=20&t=32587&start=10#p272087) we came across with this problem: 

1. When more than two vertices clicked, "PointsOnCircle" grays out
2. The PointsOnCircle can't be used for cylinder surface and an edge. 

Do we miss something?

# Attachments 
[bar.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2642033/bar.zip)

realthunder, 2018-12-04:
You need to first select a point (any element can be considered as a point, using its base coordinate), and then a circle. More points selection can be added later either during constraint creation time, or later through drag and drop into an existing constraint. 

In another word, the order of selection is important.

ceremcem, 2018-12-04:
Selecting Point, Circle, Points... works. However, results differ between creating two "PointsOnCircle" constraints and one "PointsOnCircle" constraint with multiple points: 

# Multiple Points:

[image]

# Two distinct constraints:

[image]

realthunder, 2018-12-04:
Nice catch. Fixed.

closed 2018-12-04 by realthunder

ceremcem, 2018-12-04:
Bug hunting season open! :)

commit 8bbe8e5d (2018-12-04): Constraint: allow multiple elements from the same part

Fixes #150