realthunder/FreeCAD_assembly3 #151: Unnecessary circular dependency error

ceremcem, 2018-12-07:
In [this post](https://forum.freecadweb.org/viewtopic.php?f=3&t=30676&start=10#p254327) we can see the achievement is quite possible and logical. Here is the re-do the same thing with virtual constraints which produces a cyclic dependency error: 

[image]

# Attachments 

[dynamic-dimensions-error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2656379/dynamic-dimensions-error.zip)

realthunder, 2018-12-07:
Which revision are you using? Have you sync both asm3 and LinkStage3? It works for me here.

[image]

ceremcem, 2018-12-07:
Ok, I'll take a snapshot of the entire VM. However, this leads us to a different issue topic: https://github.com/realthunder/FreeCAD_assembly3/issues/153

ceremcem, 2018-12-07:
Sorry, I misunderstood this: You say I could open my current files with new version but once it opened with new version of FC, it can't be opened with an older version. I interpreted it like FC 0.16 to 0.17 transition.

realthunder, 2018-12-07:
Right... I was in the middle of replying to #153. I'll never break backward compatibility. Even if there will be a future assembly WB rewritten in C++ by me, I'll definitely provide an importer for the old files.

There is no guarantee for the forward compatibility, though. As a rule of thumb, for an end user, is to always backup old file when opening it in new software.