realthunder/FreeCAD_assembly3 #154: segfault on Close All

J-Dunn, 2018-12-07:
If I used Close All , I seem to consistently get a seg fault.  Closing each file in turn seems fine. 

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.Unknown
Build type: Unknown
Python version: 2.7.15
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x385c0) [0x7fb34aa305c0]
#1  /usr/lib64/dri/nouveau_dri.so(+0x59f1b9) [0x7fb33dab31b9]
#2  /usr/lib64/dri/nouveau_dri.so(+0x46ff4e) [0x7fb33d983f4e]
#3  /usr/lib64/dri/nouveau_dri.so(+0x2ade6a) [0x7fb33d7c1e6a]
#4  /usr/lib64/dri/nouveau_dri.so(+0x271186) [0x7fb33d785186]
#5  /usr/lib64/dri/nouveau_dri.so(+0x2c60fe) [0x7fb33d7da0fe]
#6  /usr/lib64/dri/nouveau_dri.so(+0x1d8720) [0x7fb33d6ec720]
#7  /usr/lib64/dri/nouveau_dri.so(+0x1d89e8) [0x7fb33d6ec9e8]
#8  /usr/lib64/dri/nouveau_dri.so(+0x1d8a73) [0x7fb33d6eca73]
#9  /usr/lib64/dri/nouveau_dri.so(+0x1d8b8d) [0x7fb33d6ecb8d]
#10  /usr/lib64/dri/nouveau_dri.so(+0x2c0ff9) [0x7fb33d7d4ff9]
#11  /usr/lib64/dri/nouveau_dri.so(+0x2c11c2) [0x7fb33d7d51c2]
#12  /usr/lib64/dri/nouveau_dri.so(+0x2c6229) [0x7fb33d7da229]
#13  /usr/lib64/dri/nouveau_dri.so(+0xbc3d8) [0x7fb33d5d03d8]
#14  /usr/lib64/dri/nouveau_dri.so(+0x27b8fd) [0x7fb33d78f8fd]
#15  /usr/lib64/dri/nouveau_dri.so(+0x4638cd) [0x7fb33d9778cd]
#16  /usr/lib64/dri/nouveau_dri.so(+0x4626d7) [0x7fb33d9766d7]
#17  /lib64/libGLX_mesa.so.0(+0x49d03) [0x7fb33e364d03]
#18  /lib64/libGLX_mesa.so.0(+0x1d3b1) [0x7fb33e3383b1]
#19  0x7fb34e68b5a9 in QGLContext::reset() from /lib64/libQtOpenGL.so.4+0x89
#20  0x7fb34e61916c in QGLContext::~QGLContext() from /lib64/libQtOpenGL.so.4+0x5c
#21  0x7fb34e6191c2 in QGLContext::~QGLContext() from /lib64/libQtOpenGL.so.4+0x12
#22  0x7fb34e612bab in QGLWidget::~QGLWidget() from /lib64/libQtOpenGL.so.4+0x3b
#23  0x7fb351af3040 in CustomGLWidget::~CustomGLWidget() from /svn/FreeCAD-LinkStage3/linkstage3_build/lib/libFreeCADGui.so+0x40
#24  0x7fb351af3062 in CustomGLWidget::~CustomGLWidget() from /svn/FreeCAD-LinkStage3/linkstage3_build/lib/libFreeCADGui.so+0x18

realthunder, 2018-12-07:
Can you please close the Start page before opening any file. And then open a file and try close all. Will it crash? Also, can you please try this in upstream FreeCAD, too?

realthunder, 2018-12-08:
For upstream, you can just manually open multiple files, and then use close all. See if you get the same crash when start page is open. I don't experience any similar crash. So I suspect it may related to your system, or more specifically, your graphics driver. I want to know if upstream has the same problem to see if it is caused any change I made.

> If I save and close the next one down now shows as change, whereas it was not previously. Is this expected?

It depends on the order you save the files. If you save the top level file, then FC will sort all dependent files are save them in correct order. However, if you save the top level file first, and then go to a sub file and save it again, the sub file timestamp will change, and when the top level file is opened again next time, it will detect that change.

J-Dunn, 2018-12-08:
I can only compare to non asm3 files and when I open these files I do not get a crash on either version. 

Is it expected that a file will be marked as modified, when there is not reason it should be changed and there is no 'undo' to be done ?

realthunder, 2018-12-08:
> I can only compare to non asm3 files and when I open these files I do not get a crash on either version.

Do you mean you didn't get crash when closing multiple non-asm3 files _while start page is open_ in both version? Do only variable is whether the file contains assembly? That's unexpected.

> Is it expected that a file will be marked as modified, when there is not reason it should be changed and there is no 'undo' to be done ?

That is expected. Because the external link is tracked by file name and time stamp only, no file hash and stuff. You should avoid manually save sub files. Always save the top level files, or use save all.

J-Dunn, 2018-12-08:
> That is expected.

Oh dear, it's not doing on the asm3 files either now. I have been adding assy wrappers to the lower level files so there may have been an issue there. I had been seeing this quite consistently until today but I don't have enought back copies ( or time ) to track it down to the state I had earlier this week. 

I suggest you drop it for now. I'll reopen this if I see it happening again.

J-Dunn, 2018-12-08:
> That is expected. Because the external link is tracked by file name and time stamp only,

How does that explain the top level assy file showing as being modified when it isn't modified?

closed 2018-12-08 by J-Dunn

realthunder, 2018-12-08:
Because it detects a time stamp change on its linked file. Imaging in swap in a different linked file but with the same name, all objects in the top level file that depends on the linked file will have to be recomputed.