realthunder/FreeCAD_assembly3 #156: inconsistent selection feedback

J-Dunn, 2018-12-08:
Minor UI issue. 

When selecting several items , eg. to apply a constraint, the first item changes colour as soon as it is selected. Subsequent items do not change when cntl-click selected, but only when the mouse pointer is moved away from the selected object. 

This is confusing since inconsistent, as unhelpful since there is not visual f/b to indicate whether the second selection has selected something or not. This feedback is valuable when trying to select vertices or edges when it is often tricky to get just the right spot. 

This is presumably easy to fix.

J-Dunn, 2018-12-08:
thanks I will look into that but it seems ( as well as one can tell from your screencast which does not reveal the mouse clicks ) that you are displaying what I reported ). 

First surface:
OnHover, the surface goes yellow;  OnClick it goes green.
subsequent surfaces:
OnHover, the surface goes yellow; OnClick no change; OnExit goes green.

I have checked and there is no visual change on selecting second object with cntl-leftMouse, neither on mouseup nor on cntl-key-up. The visual indication of the selection happens when the mouse leaves the surface.

closed 2018-12-10 by realthunder

J-Dunn, 2018-12-10:
OK, now I know that it works differently from upstream and contrary to what one would expect from most software, I can work with it but it is very confusing to new users.  I guess there was a good reason to change the priority and at least you are aware some improvement could be made maybe it can get fixed at some point. 

Fixing the priority would seem the "right" way to do this but there may be a technical reason why that is not practicable. Thanks for the explanations. I'm getting alone fine now I know how it works.

realthunder, 2018-12-10:
I think you missed fact that I closed this issue with a commit, which means it has been fixed. Please sync my LinkStage3 branch and try again.