realthunder/FreeCAD_assembly3 #158: magic moving holes

J-Dunn, 2018-12-08:
I have complex block with several holes inside an assy wrapper. This file is fine in itself.

When I include it and another assy into a parent, both show correctly initially. That is the state capture in the attached file ( which is a .tar.gz renamed to a .gz to keep MS happy ). 

Pretty much as soon as I start moving the bearing assy around to align with one of the wholes something messes with the hole in "int-plate-mtg-hole" and shifts to the other side , with its centre line on the surface and crossing one of the other holes.

I have gone over this several times and it seems that as soon as I move these two object around the block gets corrupted. The base file also gets screwed. 

/tmp/guide-block-assy-OK-collection.gz

Oh well, this drag and drop thing just ain't Linux aware I guess. I'll drop on the forum as before.

realthunder, 2018-12-09:
I didn't experience any problem with your file, although there is a missing `bearing` in file `bearing+axis+spacer`. I can only guess from your description. Did you authored the bearing-block using upstream FreeCAD? It probably has nothing to do with asm3. Just open your original bearing-block file using my branch of FreeCAD, and recompute it. Do you see the same hole shifting?

Please report the version of your authoring upstream FreeCAD, using `Help -> About FreeCAD -> Copy to clipboard`, and also the version information of my branch of FreeCAD that you compiled yourself.

BTW, a screencast can explain a bit more in many cases. You can try [peek](https://github.com/phw/peek/releases). Just download the AppImage, and make it executable by `chmod +x`, and run it. It can capture GIF. Simply drop the GIF here to attach the screencast.

J-Dunn, 2018-12-09:
Sorry for the omission, it is normal that you don't see the bug at that stage.
It seems this is the missing file, I had to rename it as zip to get it to upload. It's a fcstd ;)
[bearing-6200.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2660409/bearing-6200.zip)

If you open the top level assembly file; select the bearing-axis-spacer assembly in freeview and then grab the move tool (A,M) you will see the circular lines around that assy. At this point all is well. As soon as you move / rotate that assy with the mouse, you will notice  hole has jumped in the block object. 

This does not get recovered by cancelling, there is no undo available. You will need to restore the block assy object from the file archived.

J-Dunn, 2018-12-09:
BTW, I wanted to strip the block out of the asm3 wrapper to test it in master but this does not seem to be possible. Is there some way to copy or extract it back to a form compatible with upstream. I only have the blank, w/o the holes saved as a non asm3 file.

realthunder, 2018-12-10:
I think you probably authored this block model in older upstream FreeCAD, which uses OCC (the CAD kernel) version 7.2.0, or earlier. There is known problem of topological changes when recomputing older models in OCC 7.3.0, which is what is used to compile FreeCAD on your computer. 

My branch added the [[Topological Naming]] feature to address these kind problem. However, when opening model from upstream, it has to re-generate all topological names from scratch. It only works well if you are using the same (either 7.3.0 or before) version of OCC. Otherwise you may still get the same problem as in upstream.

J-Dunn, 2018-12-10:
I don't have a complete log of exactly what build everything was made with but some intial builds of master were done with earlier versions of some packages before I followed advice and pulled everything like Coin4, OCCT etc. I'm currently building stage3 from a git clone to ensure I get build no. on help and can update more easily. When that is done will rebuild asm3.

J-Dunn, 2018-12-10:
Topological Naming feature may explain some problems I have been having with current models. I have been having strange error with stuff about face link etc things being missing. I will have to stop flipping back to master to compare things, or only do this with a copy.

luzpaz, 2022-03-21:
@J-Dunn is this still an issue?

closed 2022-03-22 by J-Dunn