realthunder/FreeCAD_assembly3 #160: build 0.18.16272 crashing

J-Dunn, 2018-12-10:
Even more crashes than usual since new pull and build this morning. Mainly to do with opening or closing files or organising assembly windows when multiple files opened. 

It is near to unusable right now. 

This does not seem to have anything to do graphics driver. 

```
<App::Document> DocumentObject.cpp(600): Changes to partial loaded document will not be saved
<App::Document> DocumentObject.cpp(600): Changes to partial loaded document will not be saved
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Cannot recompute a partially loaded document
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x385c0) [0x7fe6f62ad5c0]
#1  0x7fe6fd0844c6 in Gui::GUIApplication::notify(QObject*, QEvent*) from /svn/linkstage3_build/lib/libFreeCADGui.so+0x196
#2  0x7fe6f6ad236f in QCoreApplication::notifyInternal(QObject*, QEvent*) from /lib64/libQtCore.so.4+0x6f
#3  0x7fe6f70a7a2b in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /lib64/libQtGui.so.4+0x18b
#4  /lib64/libQtGui.so.4(+0x3153d4) [0x7fe6f711e3d4]
#5  0x7fe6f711d000 in QApplication::x11ProcessEvent(_XEvent*) from /lib64/libQtGui.so.4+0xb50
#6  /lib64/libQtGui.so.4(+0x33b4fe) [0x7fe6f71444fe]
#7  /lib64/libglib-2.0.so.0(g_main_context_dispatch+0x15d) [0x7fe6f375026d]
#8  /lib64/libglib-2.0.so.0(+0x4e638) [0x7fe6f3750638]
#9  /lib64/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7fe6f37506d0]
#10  0x7fe6f6b00589 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQtCore.so.4+0x79
#11  /lib64/libQtGui.so.4(+0x33b68b) [0x7fe6f714468b]
#12  0x7fe6f6ad0b23 in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQtCore.so.4+0x43
#13  0x7fe6f6ad0dfe in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQtCore.so.4+0x11e
#14  0x7fe6f6ad633e in QCoreApplication::exec() from /lib64/libQtCore.so.4+0x9e
#15  0x7fe6fcf9160d in Gui::Application::runApplication() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x22c9
#16  ./FreeCAD(main+0xa17) [0x409013]
#17  /lib64/libc.so.6(__libc_start_main+0xf3) [0x7fe6f6299413]
#18  ./FreeCAD(_start+0x2e) [0x4084ce]

```
OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16272 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: a97c64b0a6e936fb6c8e138b722a105776782ac6
Python version: 2.7.15
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

realthunder, 2018-12-11:
Can you please try creating multiple files inside upstream FreeCAD? Just simple file with some simple geometries. But must be more than one file. And then do a few window operations like you did in my branch. Also try the close all function, and see if it crashes the upstream as well.

J-Dunn, 2018-12-11:
OK, I created three simple object : cylinder, slab , rect block with hole. Both versions open and close these without problem and keep track of whether Close and Close All should be greyed or not. 

As soon as I open one of my larger hierarchies of assy objects and do Close All I get a messy crash.

J-Dunn, 2018-12-11:
I have noted one anomaly which may be a clue. If I open my bearing-axis-spacer.fcstd it opens up the the dependent files one by one as expected. However, if I open it via File | Open it immediately close two of the five files.  If I open from File | "recent" or from the start page ( which is effectively "recent" ) all five stay open. 

In both cases Close All causes the crash.

J-Dunn, 2018-12-11:
I have been trying a lot of combinations and orders of opening and methods of opening ( start page , file open dlg, file recent ). 

It seems minimum conditions is one single assy object then open bearing-spacer which is a hierachy and opens a dependent file.

realthunder, 2018-12-11:
Can you please try download [this](https://github.com/realthunder/FreeCAD_assembly3/releases/download/0.9/FreeCAD-asm3-20181211-09be33b1-eeda4c8.glibc2.17-x86_64.AppImage). `chmod +x` to make it executable, and run it. You need to close your local FreeCAD before running it. Try to see if you can reproduce the problem in this build.

J-Dunn, 2018-12-11:
Thanks but our security policy won't allow installation of non-distro binaries.  Is there a branch I can clone to get source for that, or was there some point other than convenience for  proposing a binary format ?

ceremcem, 2018-12-11:
@J-Dunn the AppImage is like a shell script. You don't need to install it, you'll need to just run it.

realthunder, 2018-12-11:
Yes, you don't need to install it. Just run it. It is built from my branch which is the same as yours. However, it includes (almost) all dependencies. I just want to rule out if this crash is caused by some external library.

J-Dunn, 2018-12-12:
```
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x385c0) [0x7fef3229f5c0]
#1  /usr/lib64/dri/nouveau_dri.so(+0x59f1b9) [0x7fef1e4671b9]
#2  /usr/lib64/dri/nouveau_dri.so(+0x46ff4e) [0x7fef1e337f4e]
#3  /usr/lib64/dri/nouveau_dri.so(+0x2ade6a) [0x7fef1e175e6a]
#4  /usr/lib64/dri/nouveau_dri.so(+0x271186) [0x7fef1e139186]
#5  /usr/lib64/dri/nouveau_dri.so(+0x2c60fe) [0x7fef1e18e0fe]
#6  /usr/lib64/dri/nouveau_dri.so(+0x1d8720) [0x7fef1e0a0720]
#7  /usr/lib64/dri/nouveau_dri.so(+0x1d89e8) [0x7fef1e0a09e8]
#8  /usr/lib64/dri/nouveau_dri.so(+0x1d8a73) [0x7fef1e0a0a73]
#9  /usr/lib64/dri/nouveau_dri.so(+0x1d8b8d) [0x7fef1e0a0b8d]
#10  /usr/lib64/dri/nouveau_dri.so(+0x2c0ff9) [0x7fef1e188ff9]
#11  /usr/lib64/dri/nouveau_dri.so(+0x2c11c2) [0x7fef1e1891c2]
#12  /usr/lib64/dri/nouveau_dri.so(+0x2c6229) [0x7fef1e18e229]
#13  /usr/lib64/dri/nouveau_dri.so(+0xbc3d8) [0x7fef1df843d8]
#14  /usr/lib64/dri/nouveau_dri.so(+0x27b8fd) [0x7fef1e1438fd]
#15  /usr/lib64/dri/nouveau_dri.so(+0x4638cd) [0x7fef1e32b8cd]
#16  /usr/lib64/dri/nouveau_dri.so(+0x4626d7) [0x7fef1e32a6d7]
#17  /lib64/libGLX_mesa.so.0(+0x49d03) [0x7fef1f18fd03]
#18  /lib64/libGLX_mesa.so.0(+0x1d3b1) [0x7fef1f1633b1]
#19  0x7fef30f4759e in QGLContext::reset() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4+0x5e
#20  0x7fef30ed7cc8 in QGLContext::~QGLContext() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4+0x58
#21  0x7fef30ed7d19 in QGLContext::~QGLContext() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4+0x9
#22  0x7fef30ecf507 in QGLWidget::~QGLWidget() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4+0x37
#23  0x7fef34df4d01 in CustomGLWidget::~CustomGLWidget() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x21
#24  0x7fef327c4168 in QObjectPrivate::deleteChildren() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x88
#25  0x7fef32d2ad37 in QWidget::~QWidget() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x307
#26  0x7fef332d5935 in QGraphicsView::~QGraphicsView() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x65
#27  0x7fef34e204d6 in Gui::View3DInventorViewer::~View3DInventorViewer() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x2e6
#28  0x7fef34e20539 in Gui::View3DInventorViewer::~View3DInventorViewer() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x9
#29  0x7fef34e140b2 in Gui::View3DInventor::~View3DInventor() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x172
#30  0x7fef34e14229 in Gui::View3DInventor::~View3DInventor() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x9
#31  0x7fef327c4168 in QObjectPrivate::deleteChildren() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x88
#32  0x7fef32d2ad37 in QWidget::~QWidget() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x307
#33  0x7fef331045a9 in QMdiSubWindow::~QMdiSubWindow() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x9
#34  0x7fef327c5c58 in QObject::event(QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x288
#35  0x7fef32d2e55b in QWidget::event(QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xb1b
#36  0x7fef33104eab in QMdiSubWindow::event(QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xfb
#37  0x7fef32cdee2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#38  0x7fef32ce54a0 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x270
#39  0x7fef34bef981 in Gui::GUIApplication::notify(QObject*, QEvent*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x41
#40  0x7fef327ad4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x6d
#41  0x7fef327b0b3d in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1ed
#42  /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1aaf83) [0x7fef327daf83]
#43  /lib64/libglib-2.0.so.0(g_main_context_dispatch+0x15d) [0x7fef2ceb726d]
#44  /lib64/libglib-2.0.so.0(+0x4e638) [0x7fef2ceb7638]
#45  /lib64/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7fef2ceb76d0]
#46  0x7fef327da7a1 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x71
#47  /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26bbb6) [0x7fef32d80bb6]
#48  0x7fef327ac0af in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x2f
#49  0x7fef327ac3a5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x175
#50  0x7fef327b1b79 in QCoreApplication::exec() from /tmp/.mount_FreeCAuScGTm/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x89
#51  0x7fef34b81c91 in Gui::Application::runApplication() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1651
#52  freecad-daily(main+0x74b) [0x40355b]
#53  /lib64/libc.so.6(__libc_start_main+0xf3) [0x7fef3228b413]
#54  freecad-daily() [0x4044c7]

```
The two file minimum case stated above produces a similar crash.

realthunder, 2018-12-13:
Can you please sync my LinkStage3 branch and try again.

If it still crashes, please create two sets of files. One that causes crashes, the other set containing only simple geometry that does not crash on closeAll. Please attached files here. Thanks!

J-Dunn, 2018-12-13:
This is a tar.gz with the assy based components causing a crash 
[washer-spacer-collection.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2675584/washer-spacer-collection.zip)
Here is the collection of non asm3 objects which will open/close without probs in both master  and asm3
[non-asm3-collection.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2675600/non-asm3-collection.zip)

J-Dunn, 2018-12-13:
build 16286:
good news:
washer-spacer-collection.zip seems OK. Can't say whether that is a fix or just lack of repeatability, though this did seem to crash pretty consistently if washer-M10 was opened first. 

Now testing with a more complex assy ( one more level of hierarch ) still crashes.  Simply open the top level assy file bearing+axis+spacer.fcstd then do Close All :: crash !! 
[bearing_axis_spacer-collection.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2675878/bearing_axis_spacer-collection.zip)

console output on crash essentially as before.  :(

J-Dunn, 2018-12-14:
Hi thunder,

is there a more recent AppImage than 31 August available ?

I wanted to compare notes with a college on this crash issue but I see you have pulled the link you supplied 3 days ago to the most recent build. 

I suggested he download the 31 Aug. AppImage but it crashed just opening my 3-washer spacer assy.: the most basic thing you can do.  He also uses Fedora but is still on Fed28.

realthunder, 2018-12-14:
I'll build one soon.

realthunder, 2018-12-15:
Here is the [link](https://drive.google.com/open?id=14jqk_u0EEYnPYIVuU9JgHbewvOY3SAUB).

J-Dunn, 2018-12-15:
Disable partial loading: thanks, I did not realise this was a feature. I thought it was erratic. However, disabling did not fix the crash.  What I find odd is the lack of repeatability. Doing the same test with the two file bearing-spacer example, it seems to crash about 50% of the time.  I have done an overnight memtest86+ and that was clean.

J-Dunn, 2018-12-15:
> Google Drive can't scan this file for viruses.
> 
> FreeCAD-asm3-20181215-2c69f77b-eeda4c8.glibc2.17-x86_64.AppImage (230M) is too large for Google to scan for viruses. Would you still like to download this file?

clicking on 'download anyway' does nothing.  :(

luzpaz, 2021-02-20:
@J-Dunn can this ticket be closed ?

closed 2021-02-20 by J-Dunn