realthunder/FreeCAD_assembly3 #161: graphics crash on

J-Dunn, 2018-12-10:
this one probably is graphics related:

```
240.847832 <asm3.sys> sys_slvs.py(49): redundant constraints
0.000177 <asm3.sys> sys_slvs.py(54): dof remaining: 0
0.032878 <asm3.sys> sys_slvs.py(49): redundant constraints
0.000182 <asm3.sys> sys_slvs.py(54): dof remaining: 0
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x385c0) [0x7f948305d5c0]
#1  /usr/lib64/dri/nouveau_dri.so(+0x59f1b9) [0x7f94762801b9]
#2  /usr/lib64/dri/nouveau_dri.so(+0x46ff4e) [0x7f9476150f4e]
#3  /usr/lib64/dri/nouveau_dri.so(+0x2ade6a) [0x7f9475f8ee6a]
#4  /usr/lib64/dri/nouveau_dri.so(+0x271186) [0x7f9475f52186]
#5  /usr/lib64/dri/nouveau_dri.so(+0x2c60fe) [0x7f9475fa70fe]
#6  /usr/lib64/dri/nouveau_dri.so(+0x1d8720) [0x7f9475eb9720]
#7  /usr/lib64/dri/nouveau_dri.so(+0x1d89e8) [0x7f9475eb99e8]
#8  /usr/lib64/dri/nouveau_dri.so(+0x1d8a73) [0x7f9475eb9a73]
#9  /usr/lib64/dri/nouveau_dri.so(+0x1d8b8d) [0x7f9475eb9b8d]
#10  /usr/lib64/dri/nouveau_dri.so(+0x2c0ff9) [0x7f9475fa1ff9]
#11  /usr/lib64/dri/nouveau_dri.so(+0x2c11c2) [0x7f9475fa21c2]
#12  /usr/lib64/dri/nouveau_dri.so(+0x2c6229) [0x7f9475fa7229]
#13  /usr/lib64/dri/nouveau_dri.so(+0xbc3d8) [0x7f9475d9d3d8]
#14  /usr/lib64/dri/nouveau_dri.so(+0x27b8fd) [0x7f9475f5c8fd]
#15  /usr/lib64/dri/nouveau_dri.so(+0x4638cd) [0x7f94761448cd]
#16  /usr/lib64/dri/nouveau_dri.so(+0x4626d7) [0x7f94761436d7]
#17  /lib64/libGLX_mesa.so.0(+0x49d03) [0x7f9476b31d03]
#18  /lib64/libGLX_mesa.so.0(+0x1d3b1) [0x7f9476b053b1]
#19  0x7f9486cb85a9 in QGLContext::reset() from /lib64/libQtOpenGL.so.4+0x89
#20  0x7f9486c4616c in QGLContext::~QGLContext() from /lib64/libQtOpenGL.so.4+0x5c
#21  0x7f9486c461c2 in QGLContext::~QGLContext() from /lib64/libQtOpenGL.so.4+0x12
#22  0x7f9486c3fbab in QGLWidget::~QGLWidget() from /lib64/libQtOpenGL.so.4+0x3b
#23  0x7f948a14df6e in CustomGLWidget::~CustomGLWidget() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x40
#24  0x7f948a14df90 in CustomGLWidget::~CustomGLWidget() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x18
#25  0x7f948389a874 in QObjectPrivate::deleteChildren() from /lib64/libQtCore.so.4+0x84
#26  0x7f9483ea5d00 in QWidget::~QWidget() from /lib64/libQtGui.so.4+0x330
#27  0x7f948a14adef in SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x151
#28  0x7f948a15258d in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::~SoQTQuarterAdaptor() from /svn/linkstage3_build/lib/libFreeCADGui.so+0xbb
#29  0x7f948a184ea6 in Gui::View3DInventorViewer::~View3DInventorViewer() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x48c
#30  0x7f948a184edc in Gui::View3DInventorViewer::~View3DInventorViewer() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x18
#31  0x7f948a17be24 in Gui::View3DInventor::~View3DInventor() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x1ae
#32  0x7f948a17be8e in Gui::View3DInventor::~View3DInventor() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x18
#33  0x7f948389a874 in QObjectPrivate::deleteChildren() from /lib64/libQtCore.so.4+0x84
#34  0x7f9483ea5d00 in QWidget::~QWidget() from /lib64/libQtGui.so.4+0x330
#35  0x7f9484278802 in QMdiSubWindow::~QMdiSubWindow() from /lib64/libQtGui.so.4+0x12
#36  0x7f948389b630 in QObject::event(QEvent*) from /lib64/libQtCore.so.4+0x210
#37  0x7f9483ea650a in QWidget::event(QEvent*) from /lib64/libQtGui.so.4+0x50a
#38  0x7f94842790c3 in QMdiSubWindow::event(QEvent*) from /lib64/libQtGui.so.4+0x143
#39  0x7f9483e516c0 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQtGui.so.4+0x80
#40  0x7f9483e58122 in QApplication::notify(QObject*, QEvent*) from /lib64/libQtGui.so.4+0x1a2
#41  0x7f9489e3442f in Gui::GUIApplication::notify(QObject*, QEvent*) from /svn/linkstage3_build/lib/libFreeCADGui.so+0xff
#42  0x7f948388236f in QCoreApplication::notifyInternal(QObject*, QEvent*) from /lib64/libQtCore.so.4+0x6f
#43  0x7f9483884f85 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /lib64/libQtCore.so.4+0x195
#44  0x7f9483886375 in QCoreApplication::exec() from /lib64/libQtCore.so.4+0xd5
#45  0x7f9489d4160d in Gui::Application::runApplication() from /svn/linkstage3_build/lib/libFreeCADGui.so+0x22c9
#46  ./FreeCAD(main+0xa17) [0x409013]
#47  /lib64/libc.so.6(__libc_start_main+0xf3) [0x7f9483049413]
#48  ./FreeCAD(_start+0x2e) [0x4084ce]

```

I will try to narrow it down a bit and provide a fcstd.

[SBR20-beam-crash-collection.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2664349/SBR20-beam-crash-collection.zip)

luzpaz, 2021-02-22:
@realthunder verdict?

luzpaz, 2021-06-24:
@realthunder can you recreate?

realthunder, 2021-06-24:
No, I can't recreate it. Like the OP said, probably graphics driver related.

luzpaz, 2021-06-26:
Thanks. Then closing ticket

closed 2021-06-26 by luzpaz