realthunder/FreeCAD_assembly3 #162: feature request: Constraint toggle

J-Dunn, 2018-12-11:
Hi , I just brought this up on the forum but this may be a better place since it amounts to a feature request. 

[image]

as you can see in a complex piece the red lines marking the constraints are preventing a clear view of the model. This is obviously important info but at some stage I need to see the model , not all the construction detail.  It would be nice to have a means to toggle visibility of the constraints so as to see the model  itself in cold, grey steel.

J-Dunn, 2018-12-11:
It appears that the eyeball icon "auto element visibility" may hide this but I don't seem to be able to turn it back on again.

realthunder, 2018-12-11:
Yes, the eyeball icon automates the visibility. When it is activated, you can use mouse hovering on the constraint icon in treeview to show the constraining element. When the mouse leaves, the element will be hidden. When the eyeball is deactivated, you can manually toggle the element visibility use treeview just like other shape, by using the `space` key.

[image]

J-Dunn, 2018-12-11:
Thanks but I think this is a bit different from element visibility.  It is the visiblility of the constraints ( red lines ) which I am trying to manage. 

The first time I hit they eyeball they disappear but I don't seem to be able to make them visible again without closing the program and restarting.  

Maybe that just me not knowing how to drive this yet, so maybe you can tell me if I'm missing a trick. 

thx

realthunder, 2018-12-11:
When the eyeball is activated, it automatically hides all visible constraining elements. _However_, once the eyeball is deactivated, it _does not_ automatically reveal all constraining elements. The user have to manually go through the tree view and toggle the element visibility one by one. The rational is similar to what you have mentioned, that is, the user usually does not want to be shown _all_ constrain elements, just the one they needed.

J-Dunn, 2018-12-11:
OK, I just found it odd there was not way to recover the initial situation other than closing FreeCAD. 

I guess that has most practical needs covered, so I'll close this. Thanks for explaining the logic.

closed 2018-12-11 by J-Dunn