realthunder/FreeCAD_assembly3 #163: feature request : "no tool" icon.

J-Dunn, 2018-12-11:
One thing which seems to be missing here is a mouse method for recovering the pointer cursor ( ie no special tools in operation ).

If I'm in a move operation in asm3, with the gimble rings around the part I have to use the keyboard to get out. Since I have the mouse on the left of the keyboard, this usually means dropping the mouse and hitting Esc. then having to find the mouse again, or taking my eyes of the screen to locate the Esc key, then find my point of attention on the screen again : disruptive to workflow.

If I have hand and eye coordinated on a point on the screen it is less disruptive to follow the mouse to an icon and come back that to leave the screen with my eyes and change input device. 

How about an icon with the arrow cursor which cancels any active tool like the Esc key. There are a plethora of icons for selecting different tools but not one for selecting "no tool". At first I was searching for some such icon, thinking I must have missed it or not recognised which icon it was.

GIMP toolbox works like this.

realthunder, 2018-12-11:
Yes, I can modify the GUI as suggested. However, I would suggest you to get used to one hand mouse and one hand keyboard. As a engineering application, you are required to constantly enter numerical parameter. There are commercial CAD that are optimized for touch screen, but not FreeCAD. You'll need to use keyboard a lot. In addition, a lot of toolbar buttons have keyboard shortcut, and is much faster to get activated by keyboard if you get used to it.

J-Dunn, 2018-12-11:
I'm fine with using keyboard to enter numbers or type a label, I pretty fast at typing and use all my fingers ;) But for a single input operation it is not the most ergonomic solution. 

I agree kb shortcuts like V,F are great and better than navigating a menu hierarchy which is slower.

J-Dunn, 2018-12-11:
kb shortcuts would also handy where many common tool icons are not immediately visible because of needing to expand some toolbars. I have noted many hover hints indicate when a shortcut exists. One big time waster, I have found, is selecting the sketch tools which I nearly always need to expand to get the one I need. But I guess that is an issue for master not asm3.

realthunder, 2018-12-14:
> Proposal: Double tap to `Esc` to select the default tool.

What do you mean by 'select the default tool'?

J-Dunn, 2018-12-15:
I presume he means the arrow cursor selection tool, which is what I was referring to when I wrote about "no tool".  In fact it is a tool which allows selecting objects, surfaces, vertices etc. This is common with master. 

If you have ever used GIMP, their toolbox has a variety of image manipulation tools and the first tool in the palette is the selection tool, which has the same cursor as here.  It seems an odd omission that there is not an icon for it and that I need to change input device when working with a mouse just to get back to the selection tool.  This is a very common operation ( probably the most common in terms of selecting tools ). 

Since this is what you get when you "cancel" a tool selection, I guest that makes it the default tool.

ceremcem, 2018-12-15:
> What do you mean by 'select the default tool'?

> I presume he means the arrow cursor selection tool

Yes, I thought @J-Dunn was requesting "selection tool" as the fallback tool (default tool) so I intended to propose a keybinding for that. What I understand from a "selection tool" is a tool that 
* doesn't alter the model in anyway 
* is used to select entities for further manipulation or examining their properties 
* can select multiple entities at once by clicking and dragging to cover an area, like so: 

[image]

However, I bethought that Gimp had such a selection tool with arrow icon but I can't find such thing in Gimp 2.10. 

If I understand the request wrongly, you can simply ignore my suggestion.

J-Dunn, 2018-12-15:
Sorry, I've lost track of what this was all about. I'll close now.  If I find myself in a similar situation I'll take more specific notes of the context and reopen.

closed 2018-12-15 by J-Dunn