realthunder/FreeCAD_assembly3 #164: Issue with upgraded model

ceremcem, 2018-12-11:
I intended to close the issue https://github.com/realthunder/FreeCAD_assembly3/issues/141. I updated FC beforehand from `39c32719` to `09be33b1` regarding to https://github.com/realthunder/FreeCAD_assembly3/issues/151. 

In the first opening I saw lots of warning messages (which is normal, afaik) that says the corresponding geometry is updated from 0.9.70200.3 -> 0.10.70200.3. Everything seemed to be correct till this point. I recomputed the model and it threw the error on the Report View:

```
<Part> PartFeature.cpp(186): 16Standard_Failure invalid shape type: Pad013.;#95d1f;:G#87f36;XTR;:T4133:6:F.Face13
<Part> PartFeature.cpp(186): 16Standard_Failure invalid shape type: Pad013.;#95d4a;:G#87f36;XTR;:T4133:6:F.Face15
invalid shape type
Clear constraints to external geometry
invalid shape type
Clear constraints to external geometry
invalid shape type
Clear constraints to external geometry
invalid shape type
Clear constraints to external geometry
Exception (Tue Dec 11 18:20:22 2018): Property 'keystone_length' not found
in '<<dimensions>>.keystone_length'
in property binding 'Length'  
```

# Reproduction

1. Open the `project.zip/enclosure.fcstd` (sent privately) 
2. Recompute

realthunder, 2018-12-12:
Thanks for reporting. There is a slight change in topological mapping algorithm, but the result seems very disruptive. I'll make a change to make different versions coexist. 

BTW, I initially experience crashes when opening your file, which was caused by a recent change in partial loading code. However, I didn't hear you complain. Did you by any chance manually disabled partial loading feature?

ceremcem, 2018-12-12:
> Did you by any chance manually disabled partial loading feature

Yes, possibly after [this comment](https://github.com/realthunder/FreeCAD_assembly3/issues/62#issuecomment-418927644) of yours: 
[image]

realthunder, 2019-01-13:
BTW, you may also need this little patch in SheetMetal

```diff
diff --git a/SheetMetalCmd.py b/SheetMetalCmd.py
index 6c11f78..20c9749 100644
--- a/SheetMetalCmd.py
+++ b/SheetMetalCmd.py
@@ -205,7 +205,7 @@ def smBend(bendR = 1.0, bendA = 90.0, miterA1 =0.0,miterA2 =0.0, flipped = False
         wallSolid.translate(revFace.normalAt(0,0) * unfoldLength)
         resultSolid = resultSolid.fuse(wallSolid)
       
-  Gui.ActiveDocument.getObject(MainObject.Name).Visibility = False
+  MainObject.ViewObject.Visibility = False
   return resultSolid
   
 class SMBendWall:
```

ceremcem, 2019-01-14:
> Could you tell me your OCCT version.

```
aea@fc2:/root$ cat occt/occt71/README.md 
### OCCT 7.1.0p1
```

I synced your FC branch and Asm3. At the moment, I see only 1 error in the console:

```
<unknown exception traceback><type 'exceptions.TypeError'>: str and Base.Quantity cannot be mixed in Quantity.__sub__.
Make sure to use matching types.
```
(...which doesn't point out the actual error or doesn't contain any information useful to the user, thus it should be considered as a bug, I suppose.)

Anyway, I applied the SheetMetal patch and restarted FreeCAD, here is the new error: 

```
<unknown exception traceback><type 'exceptions.TypeError'>: str and Base.Quantity cannot be mixed in Quantity.__sub__.
Make sure to use matching types.
Rebuilding Views for: Page/base-pcb001
Rebuilding Views for: Page001/pcb
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 93, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2214, in linkSetup
    getProxy(o,AsmElement).parent = self
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 25, in getProxy
    checkType(obj,tp)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 22, in checkType
    objName(obj),tp.__name__))
<type 'exceptions.TypeError'>: Expect object enclosure#Element189 to be of type "AsmElement"
5.25087 <App> Document.cpp(2483): Failed to restore enclosure#Elements: Expect object enclosure#Element189 to be of type "AsmElement"
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 93, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2165, in linkSetup
    cstr = getProxy(o,AsmConstraint)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 25, in getProxy
    checkType(obj,tp)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 22, in checkType
    objName(obj),tp.__name__))
<type 'exceptions.TypeError'>: Expect object enclosure#Constraint159 to be of type "AsmConstraint"
5.46857 <App> Document.cpp(2483): Failed to restore enclosure#Constraints: Expect object enclosure#Constraint159 to be of type "AsmConstraint"
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 93, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2343, in linkSetup
    o.Proxy.parent = self
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'parent'
5.60755 <App> Document.cpp(2483): Failed to restore enclosure#Relations: 'NoneType' object has no attribute 'parent'
```

I restored the original project, opened again and deleted the external geometries in `Sketch039`:

[image]

...then recomputed. Closed the document while saying "Yes, save changes". Everything seemed to be fine. 

Then re-opened the same document. Lots of errors popped up: 

```
Fatal Error at file /home/aea/curr-projects/scada-gateway/case2/case-v3/enclosure.fcstd, line 46633, char 27
Invalid Document.xml: expected end of tag 'XLinkSubList'
<PropShape> PropertyTopoShape.cpp(318): Recomputation required for document 'rpi_with_env' on geo element version change: 1.9.70200.3.0 -> 1.10.70200.3.0
<PropShape> PropertyTopoShape.cpp(318): Recomputation required for document 'rpi3' on geo element version change: 1.9.70200.3.0 -> 1.10.70200.3.0
<PropShape> PropertyTopoShape.cpp(318): Recomputation required for document 'MC_1_5_2_G_3_81__1803277' on geo element version change: 1.9.70200.3.0 -> 1.10.70200.3.0
<PropShape> PropertyTopoShape.cpp(318): Recomputation required for document 'rj45_90_thd' on geo element version change: 1.9.70200.3.0 -> 1.10.70200.3.0
<unknown exception traceback><type 'exceptions.TypeError'>: str and Base.Quantity cannot be mixed in Quantity.__sub__.
Make sure to use matching types.
[... 25 lines cut ...]
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 93, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2343, in linkSetup
    o.Proxy.parent = self
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'parent'
2434.11 <App> Document.cpp(2483): Failed to restore enclosure#Relations: 'NoneType' object has no attribute 'parent'
Rebuilding Views for: Page/2mm-metal
Rebuilding Views for: Page001/Unnamed
Rebuilding Views for: Page002/Unnamed
Rebuilding Views for: Page003/Unnamed
Rebuilding Views for: Page004/Unnamed
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 167, in claimChildren
    return self.ViewObject.Object.Group
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Group'
Traceback (most recent call last):
  File "<string>", line 2, in <module>
<type 'exceptions.IOError'>: 'FeaturePython' object has no attribute 'Group'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 26, in <module>
<type 'exceptions.IOError'>: 'FeaturePython' object has no attribute 'Group'
```

What should I do next?

realthunder, 2019-01-14:
Wow, that's a serious bug. Thanks for reporting! I have tried to migrate your document many many times, but never bother to do save and restore. 

Please sync LinkStage3 and asm3 again. I have tried it and it now restore fine. But there are still objects need to recompute after migrate and restore. I'll look into it further.

ceremcem, 2019-01-14:
Currently opening the original project, deleting the external geometries in Sketch039, recomputing results everything fine (just like before the last update). Saving and re-opening gives the original error: 

```
<unknown exception traceback><type 'exceptions.TypeError'>: str and Base.Quantity cannot be mixed in Quantity.__sub__.
Make sure to use matching types.
```

ceremcem, 2019-01-16:
Does everything work smoothly on your side?

realthunder, 2019-01-16:
Not really. Discovered a few more problems. I must say your document here is a really good test subject. It exposes many problems including the fact that current FC has trouble handling documents with 1K+ objects. I have made an improvement to the expression completer that solves one of these problems. You will find that editing spreadsheet in your enclosure document a lot faster.

ceremcem, 2019-01-16:
Okay. I think that document should be ([has to be?](https://github.com/realthunder/FreeCAD_assembly3/issues/151#issuecomment-445169133)) able to be upgraded without any manual modification when this issue is resolved. Till that point, I can use the previous VM for that document. 

I'm beginning a new project. Do you recommend using the current version or the version that the test document works?

realthunder, 2019-01-16:
Using the new version for new project of course. And if you can, use a new VM with OCCT 7.3, and do not mix files with earlier version. I'll explain why. And sure, as a developer the aim should be to make the migration as smooth as possible, but that is not always achievable, especially with complex things like a CAD file. 

For the jumping external geometry problem you experienced here, the actual culprit is OCCT. With version before 7.3, there is something inherently unstable in topological naming, probably due to boundary situation according to my experience, e.g. fusion or cutting with two solids that have some faces overlap. When I recomputed your document, probably one out of four times there is no need to do any manual modification, i.e. the OCCT topological names just stay the same. With 7.3, the name seems a lot more stable. However, the price to pay is that the name generated by 7.3 is incompatible with earlier version.

My new topo name on top of OCCT's plain index based name is supposed to help with this transition, in the sense that it can track the OCCT name change. But it's not bulletproof, and the design goal is never to make things fully automatic, but instead, to detect change and prefer to bring user's attention to do manual correction. On top of that, there is another complication. My topo naming algorithm is also evolving. It is unfortunate that you are caught in a bad version starting from last Sep, and is only fixed in  Nov.

The upstream FC is somewhat less affected by OCCT version change, simply because it does not track its version number. As long as you don't perform full recomputation, the topo name stays the same. But this is like a time bomb waiting for disaster to happen. And it is especially so for your document, where lots of stuff is linked by expressions.