realthunder/FreeCAD_assembly3 #165: more nouveau crashing

J-Dunn, 2018-12-19:
this may be a simpler example: single file rectangular sketch with vert + horiz length constraints. 

Open Part-designer
Go into sketch editor 
Click on horizontal length constraint to edit it ....  bang ! 

No more FreeCAD.

ceremcem, 2018-12-19:
I can confirm that the crash is happening with above steps. 

# Traceback 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f8d23003060]
#1  0x7f8d23930bb3 in std::_Rb_tree_increment(std::_Rb_tree_node_base const*) from /usr/lib/x86_64-linux-gnu/libstdc++.so.6+0x13
#2  0x7f8cf4f091e2 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /opt/FreeCAD/lib/SketcherGui.so+0xee2
#3  0x7f8d2cc72909 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x2d9
#4  0x7f8d2a2a84fb in SoEventCallback::handleEvent(SoHandleEventAction*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0xcb
#5  0x7f8d2a2bdba0 in SoNode::handleEventS(SoAction*, SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x90
#6  0x7f8d2a061624 in SoAction::traverse(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x64
#7  0x7f8d2a1c80ee in SoChildList::traverse(SoAction*, int, int) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x26e
#8  0x7f8d2a2b21b6 in SoGroup::doAction(SoAction*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x56
#9  0x7f8d2a2d1857 in SoSeparator::doAction(SoAction*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x27
#10  0x7f8d2cb92ae0 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x80
#11  0x7f8d2a2bdba0 in SoNode::handleEventS(SoAction*, SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x90
#12  0x7f8d2a061624 in SoAction::traverse(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x64
#13  0x7f8d2a1c80ee in SoChildList::traverse(SoAction*, int, int) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x26e
#14  0x7f8d2a2b21b6 in SoGroup::doAction(SoAction*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x56
#15  0x7f8d2a2d1857 in SoSeparator::doAction(SoAction*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x27
#16  0x7f8d2a2bdba0 in SoNode::handleEventS(SoAction*, SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x90
#17  0x7f8d2a061624 in SoAction::traverse(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x64
#18  0x7f8d2a06cb7a in SoHandleEventAction::beginTraversal(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x7a
#19  0x7f8d2a062e72 in SoAction::apply(SoNode*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x472
#20  0x7f8d2a1e7a19 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x89
#21  0x7f8d2cc136fe in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1e
#22  0x7f8d2cc1b056 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x46
#23  0x7f8d2cc46899 in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x9
#24  0x7f8d2cc2b730 in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x270
#25  0x7f8d2cc24218 in Gui::NavigationStyle::processEvent(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x118
#26  0x7f8d2cc4684b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x8b
#27  0x7f8d2cc0de35 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x165
#28  0x7f8d247360a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#29  0x7f8d251fc52c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#30  0x7f8d25204ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#31  0x7f8d2c99c271 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#32  0x7f8d24735f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#33  0x7f8d25202ccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#34  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f8d2527e1a9]
#35  0x7f8d2527cb5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#36  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7f8d252a6502]
#37  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f8d1bccb7f7]
#38  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f8d1bccba60]
#39  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f8d1bccbb0c]
#40  0x7f8d24766854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#41  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f8d252a65d6]
#42  0x7f8d247347ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#43  0x7f8d24734b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#44  0x7f8d2473abd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#45  0x7f8d2c9197ea in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x18da
#46  freecad-git(main+0x862) [0x55d696b397b2]
#47  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f8d22ff02e1]
#48  freecad-git(_start+0x2a) [0x55d696b3aa0a]
```

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16279 (Git)
Build type: Release
Branch: LinkStage3
Hash: 09be33b156524599ce9a1c54dc2699bd37b55bb7
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

J-Dunn, 2018-12-19:
Thanks ceremcem. 

It seems that just about any click on a constraint in this sketch crashes, I added a line and an angle of 135 deg , wanted to delete it and just trying to select it brought down the whole thing like a lead Zepplin.

realthunder, 2018-12-20:
The crash has nothing to do with your nouveau. I am sure what those `nouveau: 	0x00000000` stuffs mean before the crash stack. The crash is actually caused by my fix of your `Preselection highlight priority` problem. Anyway, it should be fixed now.

J-Dunn, 2018-12-20:
thanks, what do you mean by now? 
I hope you don't mean I've just wasted the last 4h building stage3 for nothing?  
Is this just asm3 which you are referring to ?

realthunder, 2018-12-20:
I mean I just made a commit to fix this problem. You can sync my branch and the rebuild won't take very long time.

J-Dunn, 2018-12-20:
thanks, can you clarify whether you are talking about building FreeCAD or asm3 please. I just sync'd with stage3 and it a few small changes but they seem low enough to have triggered a build from about 12% mark.  

I'll have to have a look in the morning.

realthunder, 2018-12-20:
The LinkStage3 branch. asm3 almost never needs to rebuild.

J-Dunn, 2018-12-20:
Thanks. Gives me an excuse to go to bed ;)