realthunder/FreeCAD_assembly3 #168: feature request: a means to clear report panel contents

J-Dunn, 2018-12-22:
The report window can get a bit confusing. When doing a re-computation of an assy to see whether a problem has been resolved, it is not always clear whether the same error has been reprinted ( how many of the errors were there , which line is new ? ). 

It would be helpful to have a way to wipe the report panel clean and re-compute on a clear slate to see what problems remain. 

Maybe a new icon next to the diamond and close icon could clear the panel.

Also it seems that the panel jumps to the top after recompute , rather than showing the last line of output and the bottom. This is disorienting and requires a manual scroll.

realthunder, 2018-12-22:
Right click the report panel and select `Clear`.

The panel will activate auto scroll by default, which is deactivated if you select anything in the middle. To activate auto scroll again, move your cursor to the bottom.

J-Dunn, 2018-12-22:
> Right click the report panel and select Clear.
Ah, thanks. 
One effect I have seen here is that it seems to be attempting to clear the console output too ? 
I cleared report panel then exit FreeCAD and I had an  almost blank screen with a little bit red chaff. Like the red lettering of errors had been over written but a few bits got missed.

I may want to declutter the display , but I don't think that means I want other records of errors removed too.

closed 2019-01-21 by realthunder