realthunder/FreeCAD_assembly3 #169: Save_As dlg initialisation

J-Dunn, 2018-12-24:
It would be helpful if the Save-As dialogue was primed with the existing filename and path instead of a blank file name and the path being the ~/bin directory where FreeCAD was run from. 

It would seem to be a likely starting point that the file name for a save-as will be related to the existing filename ( eg. assy-file2.fcstd or assy-file.bak.fcstd ) and that it will be going into the same directory ( or maybe subdirectory ) of the existing copy. 

Neither am I sure that the clicking on a file name to select should result in an immediate, fake click on the Save button.  The behaviour I would expect in this case is that clicking on a file name in the list would copy that file name to filename edit control and wait for further input, which many be a modification to the file name text or a click on the Save button. 

It would seem rather unlikely that I would intend to stomp on another file while saving a copy of an existing one.  If find these kind of "we know what you want to do next, so we'll do it for you" features generally waste time rather than save it. 

In this case you never save time because you need to read and respond to the "are you sure" dialogue. Either you did intend to replace and you need to say yes, or you did not intend to overwrite and you need to cancel and start again. 

If I did need to replace an existing file, the most efficient way would be to click on the file name and then press save.  If this was an explicit user action , you could probably skip the windozy " are you sure" as well. 

Save-as probably means I want a new name derived from what I already have. Please make this as fluid as possible by filling the dialogue with the existing path and filename and let the me add my "v2" or whatever and hit Save.  I think this would be generally expected behaviour in this context. 

thx.

closed 2018-12-24 by realthunder

J-Dunn, 2018-12-24:
I synced stage3 but I only seem to get the path set. Filename still empty and one click on an existing fn still goes straight to overwrite with the "are you sure".

realthunder, 2018-12-24:
Can you please quote the 'are you sure' message in exact, so that I can search the code for it. On my ubuntu it behaves just like you wanted. Looks like a system thing in Qt. Maybe some settings on your system UI regarding double/single click?

J-Dunn, 2018-12-25:
If I do Save-As on a file ali-extru-crasher.fcstd the save dialogue comes up and the filename editbox is still empty. If I click on a file name in the list of files in the save dialogue, the filename editbox is filled with the selected filename and I get : 

> ali-extru-crasher.fcstd already exists.
> Do you want to replace it?

ie it has gone straight to actually trying to save. I do not see this kind of behaviour on any other software running on this system. I should point out that I use Fedora LXDE desktop, I have both gtk and qt based software. I just checked with K3B which is  a KDE component which uses qt AFAIK: no automatic saving.

J-Dunn, 2018-12-25:
Yes, thank you. Since I do not use KDE desktop I had not had reason to go poking around in there. I have not seen this kind of issue before but it seems they have been rearranging the Matrix recently. ;) 

Anyway, having changed that to double click, I now get the behaviour I was expecting from FreeCAD. I had assumed that this was an option which was set by software when opening the dialogue. Thanks for pointing me to a fix.  By habit I was always clicking on file names and then having to cancel. Most annoying.