realthunder/FreeCAD_assembly3 #170: Disabling scroll-to-change on numerical inputs

ceremcem, 2018-12-28:
As a continuation of https://github.com/realthunder/FreeCAD_assembly3/issues/84, Path Workbench still needs this improvement: 

[image]

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16307 (Git)
Build type: Release
Branch: LinkStage3
Hash: 0f2d4bf2f756a6dfdce49581617d2187d5f0ed22
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

```

realthunder, 2018-12-29:
Path GUI is mostly done in Python. I am not sure if this can be fixed like the C++. I'll take a look when I got time.

ceremcem, 2021-02-14:
Currently we have to click to the dialog first and only then scrolling works as normal. Resolved.

closed 2021-02-14 by ceremcem