realthunder/FreeCAD_assembly3 #177: Possible to get instance name into the spreadsheet? (Class/instance approach)

ceremcem, 2019-01-26:
Is it currently possible to get the instance name (the name we give when we create links to a part) into the spreadsheet? 

# Use case
I could parse the instance name with Regex and create the appropriate parameters for that instance.

realthunder, 2019-01-26:
You mean the label of the link or the internal name? The label property is called `Label` and the internal name is `Name`. Maybe you can give an example file for better illustration.

ceremcem, 2019-01-27:
Let's assume that we need two bolts, one is 3mm diameter and the other is 5mm diameter: 

[image]

If we could get the treeview names `3` and `5` inside the `<<bolts>>#params` sheet, we could set the each instance's diameters based on those names:

[image]

[image]

realthunder, 2019-01-30:
I don't think it is possible to achieve what you have in mind with `Link`, as it does not duplicate the geometry shape. It only allows to you to display the shape at different locations. In you case, you would have to create real object for different types of screws.

I imagine that it should be possible to use spreadsheet to generate parametric screw objects, as prototypes. You can even make it into some kind of library of screw making recipes. And then use may be a separate spreadsheet to manage instances of those screws in an assembly. The `instance` here can be a `Link`. The `prototype` must be real object with geometry.

ceremcem, 2019-04-02:
Already working approach: 
* Use expression engine to obtain document name and parse it to get the variant parameters, like here: https://github.com/realthunder/FreeCAD_assembly3/issues/202#issuecomment-478326734
* Keep the edge version of the model in a folder
* Generate the variants automatically by using a text file 
* Exclude the generated models from the Git history.

closed 2019-04-02 by ceremcem