realthunder/FreeCAD_assembly3 #182: Techdraw page is "still touched after recompute"

ceremcem, 2019-01-27:
A simple Techdraw page is giving "still touched after recompute" error, which I think due to the subshape binder usage in the target part: 

[image]

# Attachments 

[case-20190127T1410-still-touched-after-recompute.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2799973/case-20190127T1410-still-touched-after-recompute.zip)

ceremcem, 2019-01-28:
I'm nearly certain that the desired target contains no cyclic dependency requirement theoretically. 

Here is what goes on currently: 
* PartA is constrained to PartB
* Independent parts of Bolt1, Bolt2, Bolt3, Bolt4  and a socket from PartB (PartB.Socket) is fused. 
* PartA is cut with Fuse to generate desired PartC

Everything seems logical.

realthunder, 2019-01-29:
The cycle is caused by a bug in TechDraw. I'll fix it soon.

closed 2019-01-29 by realthunder

realthunder, 2019-01-29:
The problem is fixed. However, for existing files, you'll have to fix it manually.

[image]

ceremcem, 2019-01-29:
> you'll have to fix it manually

No problem. 

Works great, thanks!