realthunder/FreeCAD_assembly3 #183: toponaming partdesign: edge is lost by adding circle

dulouie, 2019-01-27:
Hello realthunder, thank you for the new release!
I'm testing partdesign toponaming and found a bug. Open the attached file with LinkStage3 and modify the sketch of the pad. Create a circle to the right of the smaller square.
Now the fillet fail, invalid edge.
When i do the same from scratch with upstream master, it works.
[toponaming.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2800003/toponaming.zip)

realthunder, 2019-01-28:
This is not really a bug of Topological Naming. But your test case does expose some GUI issue, which I have fixed. The missing element is due to sketch geometry change. The upstream has no problem because it happens that the edge index is not changed in this particular case. 

After fix, the GUI will now show missing edges with question mark when double click the fillet for editing. you'll have to manually fix the references. The proper way of doing it is to create a Pocket and reorder it. You can find out more details in this [video](https://youtu.be/cNyYg1QwTRw). Please sync my branch before trying.

dulouie, 2019-01-29:
When i have to fix all references manually after base sketch modification, the useability decreases massively. Befor your toponaming effort i had to fix the lost edge manually and after fix it too? I think your toponaming effort is exactly to avoid this behaviour? The edge has the same name but only the index number has changed. Is there any chance to implement a algorithm who fix that issue? Like Edge23 (index4)-> Edge?23 (index5)

I don't want to create a new sketch on the top face and make a pocket. This circle is representative for all parametric afterwards sketch modifications. And when you create a bigger solid you have those later sketch modification. At the moment freecad is not able to do this and i had hoped your toponaming fix it.

A similar behaviour is, when you have external references and the base sketch changed by adding a circle on the right side.
[toponaming2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2806907/toponaming2.zip)
I see you have improved the external reference tool (fix reference) for that. Can you add this "fix reference" to the TaskDressUpParameters?

realthunder, 2019-01-29:
> When i have to fix all references manually after base sketch modification, the useability decreases massively. Befor your toponaming effort i had to fix the lost edge manually and after fix it too? I think your toponaming effort is exactly to avoid this behaviour?

The first priority of my topo naming is to detect geometry change, and ask for user attention when that happens. With upstream, the change happens silently, which is a lot worse.

After that, various heuristics are performed to try to search for related element in case of missing geometry. For PartDesign dressup feature, it does just that the moment you double click it for editing, which is exactly the same as what "fix reference" does in sketch external reference.

I can continue to improve this heuristic search, but there will be no guarantee that it can correctly find the missing elements. My topo name is just way to encode model history reported by OCC, like which edge generates (or modified into) which edge or face. In your particular case, the sketch extrudes to a solid, and your fillet is on the top generated face. What OCC reports is that the sketch edge generate side face, and the edges of side face generates those top edges. When tracing the origin of those top edges, I have to stop when the element type changes. Like in your case, the top edge is generated from the side face. The trace stop there, because otherwise almost any elements can be traced back to some sketch edge, cause lots of false positive. I can probably add an option to let user perform extensive search across element types.

dulouie, 2019-01-29:
Do you know how other parasolid based cad kernels handle this?
Look at this example on onshape, it is the same as toponaming2.zip. Create a new circle in the base sketch, the fillet and the second sketch2 with reference to the circle1(defining) of the base sketch1 are solved successfully.
https://cad.onshape.com/documents/d05fab6c25895e1dab939eac/w/9bf8504465da01e903e952ec/e/42f8ae47da83050f0cb9892b

realthunder, 2019-01-30:
No, I certainly do not know how they did it. But I would guess the principle is more or less the same. It is just that extrusion operation performed by Pad in FreeCAD does not have enough information to allow tracing top edges to bottom sketches when there are new geometry added to the sketch. But I'll play around it and see if I can improve it.

dulouie, 2019-01-30:
Yeah i see your point and hope you can improve those heuristic search. My third example shows the weaknesses of the heuristic. Make a circle on the right side again and try to fix the second sketch with "Fix references". Now the arrangement of the squares is a mess.
This topic has a high priority for me because it currently prevents me from using freecad in production.
Btw: Thank you for the improvement of external references in sketcher. The defining geometry feature is very useful for me. The other effort was probably canceled: https://github.com/FreeCAD/FreeCAD/pull/1421

realthunder, 2019-01-30:
I have made some adjustment to the topo naming algo. Since you seem very eager to try, please checkout branch LinkDev and test it. And frankly, such complex feature really lacks some thorough test.

It's a quick fix, so I didn't spend any effort on backward compatibility. Once you open existing file and hit recompute, some feature will break. Please manually fix the fillet edges and try again. Or you can just recreate your test files from scratch.

dulouie, 2019-01-30:
Wow! It just works! I'm so happy thank you. I done a harder test and it is still working.
[image]

realthunder, 2019-01-30:
Nice! Please test it more, but also remember not to save the result, as I still need to tweak a bit and may break something saved at this stage. Please wait till I merge this to LinkStage3.

dulouie, 2019-01-31:
With the "refine topo naming algo" commit, my test case fails again.
`PositionBySupport: AttachEngine3D: subshape not found Chamfer.?Face8
missing shape element: ?Face8`
[toponaming4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2817057/toponaming4.zip)

realthunder, 2019-02-01:
More changes. Please check. The names should be more stable now.

[image]

dulouie, 2019-02-01:
At a first quick test i can confirm, all is working very well. I will do some more testing in the near future and report here.