realthunder/FreeCAD_assembly3 #184: AsmConstraint object has no attribute "version"

ceremcem, 2019-01-28:
After I pulled the updates for https://github.com/realthunder/FreeCAD_assembly3/issues/180, opening the same project produced the following error: 

```
0.268843 <App> Document.cpp(2727): The graph must be a DAG.
Rebuilding Views for: Page001/pcb
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1819, in execute
    if not self.version.update(self.childVersion()):
<type 'exceptions.AttributeError'>: 'AsmConstraint' object has no attribute 'version'
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1819, in execute
    if not self.version.update(self.childVersion()):
<type 'exceptions.AttributeError'>: 'AsmConstraint' object has no attribute 'version'
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1819, in execute
    if not self.version.update(self.childVersion()):
<type 'exceptions.AttributeError'>: 'AsmConstraint' object has no attribute 'version'
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1819, in execute
    if not self.version.update(self.childVersion()):
<type 'exceptions.AttributeError'>: 'AsmConstraint' object has no attribute 'version'
```

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16421 (Git)
Build type: Release
Branch: LinkStage3
Hash: f218dfb4d1e83f6086b884c3d2f3581aabd3dba1
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2019-01-28:
I can not reproduce this problem. I tried to disable relevant constraints, tried to delete the suspicious `Cut` and `Fusion` and it disappeared.

closed 2019-01-28 by ceremcem

realthunder, 2019-01-28:
This is a side effect of document with cycles. Because asm3 requires the file to be restored in dependency order, if there are cycles, the order is somewhat random.