realthunder/FreeCAD_assembly3 #189: LinkStage3 design consideration: Storing dependency arguments

ceremcem, 2019-01-29:
AFAIU, some arguments of a dependency (a sub assembly) is stored *inside* the dependency. For example, I'm using a `bolt` and `moduleA`  inside `the-assembly`. When I created a `PlaneCoincident`, between "under the head of bolt" and "a hole on moduleA", the constraint is stored both inside `the-assembly` and `moduleA`, right?

realthunder, 2019-01-29:
No. The constraint is only stored in `the-assembly`. The element used by the constraint is stored in both `moduleA` and `assembly`, so that if the element in `moduleA` is accidently deleted, you still have visual clue inside `the-assembly`.

ceremcem, 2019-01-29:
> No. The constraint is only stored in the-assembly.

Sorry, this is my mistake. It turns out that I had a hidden geometry which mislead me. 

> so that if the element in moduleA is accidently deleted, you still have visual clue inside the-assembly

This feature is provided by the element located inside `the-assembly`, right? If so, what is the purpose of the element copy stored inside `moduleA`?

My point is that, the fingerprint of a dependency shouldn't be modified regardless of its usage in various projects. Is it already working this way, or am I missing something?

realthunder, 2019-01-30:
> This feature is provided by the element located inside `the-assembly`, right? If so, what is the purpose of the element copy stored inside `moduleA`?

The element inside `moduleA` servers as the declared interface of that sub-assembly, which may be used in more than one place. Imaging you used `moduleA` in some `other-assembly`, and delete one of its element for some reason. And when you load `the-assembly` later, probably after some day, you may have a hard time remembering which element you have deleted.

Besides, it also enables swapping different sub-assemblies with the same interface. 

> My point is that, the fingerprint of a dependency shouldn't be modified regardless of its usage in various projects

I don't quite get what you mean here. There is no mechanism to lock a sub-assembly interface. You can however, lock (freeze) an upper assembly, which makes it unaffected by any sub-assembly changes.

ceremcem, 2019-02-10:
Let me clarify the origin of this issue: I thought I saw a suddenly appeared constraint inside `moduleA` which was actually added in `the-assembly` (in other words, adding a constraint that includes an element of `moduleA` inside `the-assembly` changed file contents of `moduleA`). However, it was my mistake as it turns out I just hid a geometry inside `moduleA` and that geometry is way similar to a geometry inside `the-assembly`. That's why this issue should be closed with my apologies. 

Thanks for the explanation of the purpose of the element copy located inside subassembly. I think I should understand what an element means under the hood (inside fcstd file) before commenting on this matter. 

> I don't quite get what you mean here

1. Create a assembly (`moduleA`). Fingerprint (eg. MD5) of `moduleA.fcstd` is now `abcd1234`. 
2. Create another assembly (`the-assembly`) and use `moduleA` inside it (create some constraints related with `moduleA`, for example). 

If the fingerprint of `moduleA.fcstd` becomes `defg5678` (assume that timestamp is not changed in fcstd file) just because we created some constraints related with `moduleA` inside `the-assembly`, then "Huston, we have a problem".

closed 2019-02-10 by ceremcem