realthunder/FreeCAD_assembly3 #190: How to create an Element from a bounding box face/edge/vertex?

ceremcem, 2019-01-29:
I need to create a constraint to make the pins (array of cylinders) perpendicular to the right edge (which is currently ~30°): 
[image]

If I were able to create a `MultiParallel` constraint between the closest face of the PCB (according to the screenshot) and the largest face of bounding box of the pin array, everything would be set in one shot.

realthunder, 2019-01-30:
I am not entirely sure about your intention here. When you add an array directly to an assembly, each individual array element is treated as a separate part and can be constrained independently. If you want to constrain the entire array as a single part, just wrap it inside a sub-assembly, and add that into the assembly instead.

ceremcem, 2019-02-10:
> If you want to constrain the entire array as a single part

Yes, that was the intention. However, I don't quite get how encapsulating the model inside an assembly container would help creating that simpler constraint. 

# Problem

The problem arises whenever a geometry (a model) has no suitable planar surface that you could create a constraint with. 

# Example 

Think that you have an ellipsoid which you want to create a constraint to make its longest cord line parallel to a some simple surface: 

[image]

If we could use the bounding box (the dashed virtual box) faces as if they were actual geometries, we could perform the task with some simple PointToPlaneDistance and MultiParallel constraints:

[image]

# Problem

**However**, the bounding box is a dynamic concept that we have almost no control over it. It would be changed unintentionally when some other geometry is added to the assembly. So please ignore the proposal.

closed 2019-02-10 by ceremcem