realthunder/FreeCAD_assembly3 #191: Mass updating sub-shape binders

ceremcem, 2019-01-29:
According to `Fixing broken binder` section in [my notes](https://github.com/ceremcem/freecad-notes), we need to select the binder and the bound object, then click "Sub-shape binder" button to update the binder. 

The problem is that, in a large tree, it's very hard to find the bound object manually. Right-clicking the binder "Select bound object" doesn't work for me at the moment. Even if it worked correctly, it's still hard to check every binder in a large tree because there is no notification system available for the binders that are out of sync. 

Isn't it possible to have a utility that just updates all out-of-sync binders one by one while asking with "yes-no" prompt?

ceremcem, 2019-02-08:
(Previous attempt: https://github.com/realthunder/FreeCAD_assembly3/issues/89)

ceremcem, 2021-02-14:
@realthunder I think this is already implemented, right?

realthunder, 2021-02-15:
Kind of, as the `SubShapeBinder` will be marked as error if it fails to get any bound object.

closed 2021-02-15 by ceremcem