realthunder/FreeCAD_assembly3 #192: part design one-solid rule

dulouie, 2019-02-03:
Another major topic is the one-solid rule in part design. You are the only one i know, with the necessary freecad knowledge and coding skills. Maybe you like to have a look?
https://github.com/FreeCAD/FreeCAD/pull/1398
In this discussion all agree to drop this rule but nobody wants to code it.
[image]
It is one of a last steps to a very well matured part design experience.

realthunder, 2019-02-05:
I just add a quick fix to lift the single solid restriction in `LinkDev` branch. Each body now has a property called `SingleSolid`. Set it to `False` to allow multiple solids. You can add a boolean parameter with the same name at `BaseApp/Preferences/Mod/PartDesign` using parameter editor to set the default value of this property in a new body.

Again, this is just a quick fix. It seems that newer OCC version has good support of boolean operation with multiple solids. I don't even need to add any special code for handling that. Please test it more and let me know the result.

dulouie, 2019-02-05:
When i create a new body freecad crashs.
`Program received signal SIGSEGV, Segmentation fault.`
But thanks for your time. I'm pleased you have looked.

realthunder, 2019-02-05:
Oops! Please sync and try again.

dulouie, 2019-02-05:
The crash is fixed. But if i set the SingleSolid property to false and try to pad an sketch with two circles. It dosn't work.

realthunder, 2019-02-05:
Added. Please try again.

dulouie, 2019-02-05:
Pad and revolution working very well! Big thanks!
btw: I compiled freecad with qt5 and python3 and also slovespace bindings to py3. And i noticed that asm3 feels more faster, it could be?

dulouie, 2019-02-05:
Assembly solving feels faster but it could be subjective.

dulouie, 2019-02-05:
I've tested all part design tools:
*Padding is working
*Revolution is is working
*Loft is working
*Sweep is working
*Pocket is working
*Groove is working
*Hole is working
*SubtractivePipe is working
*SubtractiveLoft is working
*Mirror is NOT working (Shape is null)
*Linear pattern is NOT working (Shape is null)
*Polar pattern is NOT working (Shape is null)
*Multi transform is NOT working (Shape is null)
*Fillets are working
*Chamfers are working
*Draft is working
*Thinkness is NOT working
*Boolean is working
*Subshapebinder is working

dulouie, 2019-02-05:
Open toponaming4.zip from #183 and create a circle in the white space of the first sketch.
The attached sketch with the pockets break.

realthunder, 2019-02-06:
More changes. Please check.

dulouie, 2019-02-06:
All tools working good, expect the draft tool is buggy as every time.
[image]

realthunder, 2019-02-06:
Can you please attach a file showing the draft tool problem?

dulouie, 2019-02-07:
Actually there is no problem with the draft tool. I forgot the set the neutral plane.
But there is a problem with the mirror tool  (pattern tools too). I can only add the last pad or pocket. Chamfers, drafts etc. dosn't work.
[draft.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2841960/draft.zip)

realthunder, 2019-02-09:
> I can only add the last pad or pocket.

This behavior is actually by design, i.e. the transform is only applicable to additive or subtractive features like Pad and Pocket. In addition, the transform is only applied to the additional sub-shape, e.g. the extra solid that is padded/pocketed, instead of the entire solid. 

But since we now allow multiple solids, it is only natural to lift the above restriction as well. I have just implemented this. Now the behavior is that for non-additive/subtractive feature, the whole solid is transformed. If there are multiple solids, and you select a face in the 3D view, then only the solid containing that face is transformed. If you instead, select the whole object (e.g. when select in the treeview), then all solids are transformed. 

For additive/subtractive features like Pad and Pocket, the default behavior is still the same, i.e. only transform the additional sub-shape (of all solids). To change that, set `SubTransform` property to False, or uncheck the `Transform sub-feature` option in the task panel.

dulouie, 2019-02-09:
I really like your enthusiasm to fix bugs or implement new features to part design. Have you ever thought about a patreon support like https://www.patreon.com/kkremitzki? 
I think if you further improve the part design workbench there will be a lot of people who will support you. 
You implemented all the stuff i want: 
* subshapebinder for relative reference to another body
* sketcher external geometry defining
* drop the body one solid rule
* toponaming
* improve mirror tool
Partdesign has at the moment no active developers and is not finished.
Maybe you'd like to work on it? 

I'am now working on design-in-context feature based on subshapebinder, external file links and partdesign.

btw: the mirror tool is not stable `Program received signal SIGSEGV, Segmentation fault.`

realthunder, 2019-02-09:
> Have you ever thought about a patreon support like https://www.patreon.com/kkremitzki?

Yeah, I intend to. Just haven't got time around to do it yet. I am actually considering taking FreeCAD to university for funding. Still thinking how to present it. I can't put my full time on FreeCAD yet due to obvious reason, but I really want to.
 

> btw: the mirror tool is not stable `Program received signal SIGSEGV, Segmentation fault.`

Can you reproduce it? Normally when FC crash it will print out stack trace. Can you please post that?

dulouie, 2019-02-10:
I compiled it against python3 and qt5:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x41100) [0x7f496dbf3100]
#1  0x7f496f0a3601 in Py::_Unicode_Check(_object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADBase.so+0x1
#2  0x7f496f53853c in App::PropertyLinkSub::setPyObject(_object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADApp.so+0x3dc
#3  0x7f496f53cab5 in App::PropertyLinkSubList::setPyObject(_object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADApp.so+0x45
#4  0x7f496f437038 in App::DocumentObjectPy::setCustomAttributes(char const*, _object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADApp.so+0x98
#5  0x7f496f43b406 in App::DocumentObjectPy::_setattr(char const*, _object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADApp.so+0x26
#6  0x7f496f43219e in App::GeoFeaturePy::_setattr(char const*, _object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADApp.so+0x3e
#7  0x7f48f57f6fae in Part::PartFeaturePy::_setattr(char const*, _object*) from /home/arne/src/FreeCAD/build/Mod/Part/Part.so+0x3e
#8  0x7f492420259e in PartDesign::FeaturePy::_setattr(char const*, _object*) from /home/arne/src/FreeCAD/build/Mod/PartDesign/_PartDesign.so+0x3e
#9  0x7f496f10428d in Base::PyObjectBase::__setattro(_object*, _object*, _object*) from /home/arne/src/FreeCAD/build/lib/libFreeCADBase.so+0x8d
#10  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(PyObject_SetAttr+0x4f) [0x7f496ed0759f]
#11  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x43d9) [0x7f496ec7b9c9]
#12  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(+0x17ba5f) [0x7f496ec75a5f]
#13  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(PyEval_EvalCodeEx+0x3e) [0x7f496ec7651e]
#14  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(PyEval_EvalCode+0x1b) [0x7f496ec773fb]
#15  /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0(PyRun_StringFlags+0x8b) [0x7f496ec485cb]
#16  0x7f496f0cd074 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/arne/src/FreeCAD/build/lib/libFreeCADBase.so+0x64
#17  0x7f496fbb56c2 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0x82
#18  0x7f496fbb583f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0xff
#19  /home/arne/src/FreeCAD/build/Mod/PartDesign/PartDesignGui.so(+0xf1926) [0x7f48ee525926]
#20  0x7f48ee52ec7c in prepareTransformed(PartDesign::Body*, Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::function<void (App::DocumentObject*, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >)>) from /home/arne/src/FreeCAD/build/Mod/PartDesign/PartDesignGui.so+0x36c
#21  0x7f48ee52f36d in CmdPartDesignMirrored::activated(int) from /home/arne/src/FreeCAD/build/Mod/PartDesign/PartDesignGui.so+0xcd
#22  0x7f496fbb8c2a in Gui::Command::invoke(int, bool, Gui::Command::TriggerSource) from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0x1ea
#23  0x7f496e1d66db in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x76b
#24  0x7f496e5faef2 in QAction::triggered(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#25  0x7f496e5fd500 in QAction::activate(QAction::ActionEvent) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf0
#26  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x246d1d) [0x7f496e6e8d1d]
#27  0x7f496e6e8f55 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe5
#28  0x7f496e7d2caa in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
#29  0x7f496e63f7d8 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d8
#30  0x7f496e7d2d53 in QToolButton::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x33
#31  0x7f496e6014a1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
#32  0x7f496e608d28 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x458
#33  0x7f496fb88378 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0x88
#34  0x7f496e1ad499 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#35  0x7f496e608029 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a9
#36  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1b8304) [0x7f496e65a304]
#37  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bae8e) [0x7f496e65ce8e]
#38  0x7f496e6014a1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
#39  0x7f496e608ae0 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#40  0x7f496fb88378 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0x88
#41  0x7f496e1ad499 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
[... cut ...]
#46  0x7f496e1ac16b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#47  0x7f496e1b42e2 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#48  0x7f496fb18037 in Gui::Application::runApplication() from /home/arne/src/FreeCAD/build/lib/libFreeCADGui.so+0x1377
#49  ./bin/FreeCAD(main+0x6c9) [0x563393fc2809]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7f496dbd609b]
#51  ./bin/FreeCAD(_start+0x2a) [0x563393fc2b0a]
```

dulouie, 2019-02-10:
With qt4 and python2.7 it is working.
But terminal shows:
`<unknown exception traceback><type 'exceptions.IndexError'>: list index out of range`

realthunder, 2019-02-11:
The crash should be fixed now. Please try again in Py3 build.

> 9e-08 <App> Document.cpp(2759): The graph must be a DAG. 0.0213472 <App> Document.cpp(3092):
> Unnamed#Mirrored still touched after recompute

This error shows you have a dependency loop in your file. Did you set the Mirror link using script? If not then there might be a bug somewhere, can you reproduce it and send me the file?

dulouie, 2019-02-11:
It is working, but the "The graph must be a DAG" message is still there.
[mirror.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2849689/mirror.zip)

realthunder, 2019-02-11:
Fixed. I made a breaking change in all Transformed features, which is necessary to maintain backward compatibility with upstream FC.

dulouie, 2019-02-11:
Working very well:)

dulouie, 2019-02-11:
Ohh but multitransform has the DAG problem.
[multitransform.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2850960/multitransform.zip)

realthunder, 2019-02-12:
Fixed. Should be working now. You need to recreate the MultiTransform to get the fixed workflow.

dulouie, 2019-02-12:
very nice! mission complete:)

dulouie, 2019-02-14:
You have merged it to LinkStage3 nice! But on mirror tool the solid selection dosn't work anymore. All solids are mirrored.

closed 2019-02-14 by dulouie

dulouie, 2019-02-18:
After some testing with the mirror tool, only all solids are mirrored. You can not add and remove single solids.

realthunder, 2019-02-18:
You can this problem happen after a while, or it always does not work? You are supposed to add single solid by select a face of that solid. And if it is an additive/subtractive feature (e.g. Pad, Pocket), you need to set `SubTransform` to `False` in the Mirror object to transform single solid.

dulouie, 2019-02-21:
Okay that's the point. When i set `SubTransform` to `False` everything is working as expected.

cbenhagen, 2023-04-22:
@realthunder sorry to dig up this old issue. Do you know if there is a corresponding issue in the upstream repo? I haven't found one so far.

It would be so nice if this feature could be merged upstream.

cbenhagen, 2024-03-10:
There is an issue now: https://github.com/FreeCAD/FreeCAD/issues/9747