realthunder/FreeCAD_assembly3 #194: subshapebinder placement bug

dulouie, 2019-02-08:
I noticed a new placement bug by the subshapebinder.
Double click on the Binder to update it. And now the binder release to the document origin although relative is set to true.
[binderPlacementbug.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2847245/binderPlacementbug.zip)

realthunder, 2019-02-09:
That's not a bug. `Relative` property is not the only factor that affects how the placement is calculated. The link information stored in `Support` property is another factor. Your attached file does not have parent-child hierarchy information in the link, so the placement is not calculated as desired. The following screencast shows how to correct it. A few points,

* The `Show all` menu action is to review all hidden property, so as to show you the problematic `Support` property content. It is not necessary to correct the shape binder
* Normal drag and drop only add an additional geometry binding. To reset the binding, hold `CTRL` key before dropping.
* After correction, the `Support` actually stores Body.Chamfer001 which holds the necessary hierarchy information to calculate the placement.

[image]

dulouie, 2019-02-09:
okay thanks problem is solved.

closed 2019-02-09 by dulouie