realthunder/FreeCAD_assembly3 #195: Problem with measurement constraints

ceremcem, 2019-02-12:
What is done: 

1. I drew the parts, assembled them. At the beginning, the wheels are on the outside of the bow-like-part. 
2. I replaced the elements with appropriate ones to get the wheels inside of bow-like-part. 

Here is erroneous file: [mechanical-20190213T0023-with-error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2857716/mechanical-20190213T0023-with-error.zip)

[image]

realthunder, 2019-02-13:
Bug fixed. Please sync asm3 repo, and manually recompute the measurement object.

closed 2019-02-13 by realthunder

ceremcem, 2019-02-14:
Unfortunately I'm getting the same results. How can I provide any further debug information?

realthunder, 2019-02-14:
Make sure your asm3 repo has the latest [commit](https://github.com/realthunder/FreeCAD_assembly3/commits/master). Then do manual recompute like below,

[image]

ceremcem, 2019-02-14:
OK, solved. The problem was that I had synced the repo already and did "Recompute object" but mistakenly I tried to recompute the whole assembly container and then recompute the Cylinder; not the "measurement object".

commit 3b7f8dfc (2019-02-13): constraint: fix PointDistance

Fixes #195