realthunder/FreeCAD_assembly3 #196: exploding assembly3?

mandd3, 2019-02-14:
Hi,
any advice on achieving Exploded_Assembly with assembly3?

Thanks!

realthunder, 2019-02-15:
I will add this feature soon. For now, I add a simple command to make it easy to do it manually. You will need to sync my LinkStage3 branch to get it. If you are not building FC yourself, please wait till next release.

Note that the LinkGroup created in the screenshot blew has synchronized placement to the source assembly. It creates a `Link` for each part inside assembly with `LinkTransform` set to `True`, which allows the user to apply additional transformation. The `TechDraw` workbench accepts `LinkGroup`, so you can easily make explode drawing with it.

[image]

ceremcem, 2019-02-15:
A-ha! Isn't it opening the doors for ["Custom scenes"](https://github.com/realthunder/FreeCAD_assembly3/issues/186)?

realthunder, 2019-02-15:
Exactly. I am think of putting this type of feature under a new sub group of the assembly container.

dulouie, 2019-02-15:
That's real App:Link performance!  really neat

mandd3, 2019-02-16:
Will be using assembly3 from now on.
thank you realthunder. Great work.