realthunder/FreeCAD_assembly3 #197: Link Array / Multiply is broken

CyberDeck, 2019-02-16:
If I follow your wiki example to create a link array it fails with

`Traceback (most recent call last):
  File "<string>", line 595, in report
  File "/tmp/.mount_FreeCA6ZT54o/usr/lib/freecad-daily/Ext/freecad/asm3/assembly.py", line 2124, in makeMultiply
    cstr.Multiply = True
AttributeError: Attribute (Name: Multiply) error: 'object of type 'NoneType' has no len()' 
`

Even by doing it manually (setting "multiply" to true in the constraint it fails basically with the same error). 
I am using the latest linux appimage FreeCAD-asm3-20190202-9a4632f4-4664316.glibc2.17-x86_64.AppImage

realthunder, 2019-02-16:
Can you please attach a file showing the problem. Thanks

luzpaz, 2021-12-27:
@realthunder can this ticket and #341 be closed?

closed 2021-12-30 by realthunder