realthunder/FreeCAD_assembly3 #198: body multiple solid basefeature pocket confusion

dulouie, 2019-02-21:
I think this has to do with multiple solids and basefeature in partdesign.
Try to though all Pocket003. But where is the pocket?
[image]
[gears_pad_error.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2889630/gears_pad_error.zip)

dulouie, 2019-02-21:
When i make the pocket on every single solid first and fuse them later together. I get these nice shape:)
[image]

realthunder, 2019-02-23:
I don't think this is related to multi solid feature. It seems OCCT has trouble with cutting the _not so valid boolean shape_. The root cause of this problem is that the involute gear uses BSpline for the teeth, and therefore the resulting _flat_ face is not really flat to OCCT. When you fuse the two involute gears by touching their _not so flat_ faces, you actually get two solids. With upstream FC you'll probably notice this as the result will be missing one gear. To correct the problem, just make these two gears touch a bit more, say, `InvoluteGear002.Placement.Base.Z = 17.999`.

BTW, with these non-flat faces, it is not advisable to map sketch on it. You get noticable slow down, and may probably get broken shape somewhere along the way. Use datum plane instead.

Another thing, just like fusing, it is also advisable to have some leeway for pocketing as well. So the datum plane should be slightly above the gear face.

dulouie, 2019-03-02:
Okay thanks for the explanation.

closed 2019-03-02 by dulouie