realthunder/FreeCAD_assembly3 #199: Segmentation fault when deleting external geometry element

bgamari, 2019-03-08:
Attached is a design which causes FreeCAD 4664316 (the realthunder AppImage release) to crash on Linux.

To reproduce:

1. Open the design
2. Open `pump-bracket/Body/Pad002/Pump mount` in Sketcher
3. Scroll to the end of the "Elements" list in the sidebar
4. Select either `ExternalEdge1` or `ExternalEdge2`
5. Press the Delete key

FreeCAD will segmentation fault:
```
$ nix run -f .nix-overlay/ freecad-realthunder -c FreeCAD
FreeCAD 0.18, Libs: 0.18R16268 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch003.e4 missing reference: Pad001.;#11c;:G3#1c9;FUS;:T11:4:E
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch003.e6 missing reference: Pad001.;#169;:T17:4:E
Exception (Fri Mar  8 18:50:40 2019): Missing external geometry reference  
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch002.e7 missing reference: Mirrored.;#d0;:T-14:3:E
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch002.e8 missing reference: Mirrored.;#d4;:T-14:3:E
Exception (Fri Mar  8 18:50:40 2019): Missing external geometry reference  
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch003.e4 missing reference: Pad001.;#11c;:G3#1c9;FUS;:T11:4:E
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch003.e6 missing reference: Pad001.;#169;:T17:4:E
Exception (Fri Mar  8 18:50:40 2019): Missing external geometry reference  
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch002.e7 missing reference: Mirrored.;#d0;:T-14:3:E
<Sketcher> SketchObject.cpp(6503): External geometry pump_bracket#Sketch002.e8 missing reference: Mirrored.;#d4;:T-14:3:E
Exception (Fri Mar  8 18:50:40 2019): Missing external geometry reference  
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x34c20) [0x7f99199cec20]
#1  0x7f98a1f6e353 in SketcherGui::ViewProviderSketch::updateColor() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x7f3
#2  0x7f98a1f7235a in SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x7da
#3  0x7f991cbeae5f in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1f
#4  0x7f991cc02fdc in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1fc
#5  0x7f991cbf0df6 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x236
#6  0x7f991cc02fdc in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1fc
#7  0x7f991cbf2f50 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x100
#8  0x7f991cbf3ac1 in Gui::SelectionSingleton::setPreselect(char const*, char const*, char const*, float, float, float, int) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x2e1
#9  0x7f98a1f42c42 in SketcherGui::TaskSketcherElements::on_elementsWidget_itemEntered(QTreeWidgetItem*) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x5f2
#10  0x7f991a47d87a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x2da
#11  0x7f991af04310 in QTreeWidget::itemEntered(QTreeWidgetItem*, int) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x40
#12  0x7f991a47d87a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x2da
#13  0x7f991ae92625 in QAbstractItemView::entered(QModelIndex const&) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x25
#14  /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x6c96e3) [0x7f991ae9a6e3]
#15  0x7f991ae9af8f in QAbstractItemView::verticalScrollbarValueChanged(int) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xcf
#16  0x7f991a47d87a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x2da
#17  0x7f991afe956e in QAbstractSlider::valueChanged(int) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2e
#18  0x7f991aed2022 in QTreeViewPrivate::updateScrollBars() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xb2
#19  0x7f991aedb377 in QTreeView::updateGeometries() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x197
#20  0x7f991ae8e56d in QAbstractItemView::doItemsLayout() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2d
#21  0x7f991aeda995 in QTreeView::doItemsLayout() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xd5
#22  0x7f991aecd4a4 in QTreeView::resizeColumnToContents(int) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x34
#23  0x7f98a1f43ea5 in SketcherGui::TaskSketcherElements::slotElementsChanged() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x3e5
#24  0x7f98a1f90e52 in boost::signals2::detail::signal_impl<void (), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x1b2
#25  0x7f98a1f848d2 in SketcherGui::ViewProviderSketch::updateData(App::Property const*) from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/freecad-daily/lib/SketcherGui.so+0x102
#26  0x7f991c88bc86 in Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x36
[... cut ...] 86_64-linux-gnu/libQtCore.so.4+0x175
#71  0x7f991a46db79 in QCoreApplication::exec() from /home/ben/.cache/appimage-run/da21b40d5cb5adf386e4a1a9eff26e0c78166bd2335e72731db16bc51497bba8/squashfs-root/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x89
#72  0x7f991c84fd49 in Gui::Application::runApplication() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1669
#73  freecad-daily(main+0x74b) [0x40355b]
#74  /usr/lib/libc.so.6(__libc_start_main+0xee) [0x7f99199bbb5e]
#75  freecad-daily() [0x4044c7]
```

bgamari, 2019-03-08:
Design exhibiting crash. There does appear to be something wrong with the dependency graph of this design (if you have any ideas on this let me know; I'm rather stuck) but nevertheless it shouldn't crash.

[repro.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2947849/repro.zip)

realthunder, 2019-03-09:
Thanks for the report. Bug confirmed. As a temporary workaround, just move your mouse away from the element list before hitting delete key.

closed 2019-03-09 by realthunder

bgamari, 2019-03-10:
@realthunder, thank you!