realthunder/FreeCAD_assembly3 #201: Disabling file last modification time

ceremcem, 2019-03-19:
Main thread: https://forum.freecadweb.org/viewtopic.php?f=3&t=31085

What problems would disabling file last modification time while saving the file cause in the long shot?

```
diff --git a/Document.xml b/Document.xml
index f2672a6..61b8563 100644
--- a/Document.xml
+++ b/Document.xml
@@ -31,7 +31,7 @@
             <String value=""/>
         </Property>
         <Property name="LastModifiedDate" type="App::PropertyString" status="1">
-            <String value="2018-09-22T06:38:17-04:00"/>
+            <String value="2018-09-22T06:39:10-04:00"/>
         </Property>
         <Property name="License" type="App::PropertyString" status="1">
             <String value="All rights reserved"/>
```

Disabling this update is likely to solve the "what did I change in those files? o_O" type of questions when the binary files are only touched because of a recomputation.

realthunder, 2019-03-25:
How did you save the project file in git? Manually decompress it? How do you manage those added/removed property files?

I think this type of problem is best to be handled by a property integration of version control system with FC.

ceremcem, 2019-03-27:
Yes, I managed to get the diffs by manually decompressing the files and overwriting a manually created git repo. 

> I think this type of problem is best to be handled by a property integration of version control system with FC.

I agree. However, I think we could eliminate the "Nothing has really changed" situation (which covers most of the cases while dealing with models from a library) by disabling last modification time.

realthunder, 2019-03-28:
The software has to make sure the saved file is really not changed, and that can only be done by actually save every objects to file first, and then compare. And it is further complicated by the zip format FC is using, which is why I said it is better let VC to handle this, as by then it will not be using zip.

Or maybe there is some pattern here we can identify. Can you show some examples (with steps) of date only modification?

ceremcem, 2019-03-28:
> And it is further complicated by the zip format FC is using, which is why I said it is better let VC to handle this, as by then it will not be using zip.

I agree. I barely remember that we had a conversation about this topic. The zip format contains some binary documents which consumes a large space, so it also isn't really suitable for being handled with a VC (at least with Git). 

> Can you show some examples (with steps) of date only modification?

1. Create a new project 
2. Add a cube and a spreadsheet 
3. Save (and unzip the fcstd file to a git tracked folder, make a commit)
4. Right click -> Mark to recompute 
5. Recompute 
6. Project file is touched.
7. Save (and unzip the fcstd file to the git tracked folder, display the diff)