realthunder/FreeCAD_assembly3 #202: A simpler form of "Expression and Spreadsheet" tutorial is needed

ceremcem, 2019-03-19:
Current demonstration (https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#demonstration) shows a full blown example for a real-world problem. However, I'm having hard times to get up and running with expression engine mode. 

Currently I'm trying to 
* Get the document name 
* Use regex to extract a number from the name 
* Assign this number to the cell (or multiple values to multiple cells)

I'm able to achieve the number by the following Python code: 

```py
import re 

doc_name = App.ActiveDocument.Label
regex_pattern = re.compile("\d+")
regex_match = regex_pattern.search(doc_name)
foo = 1
if regex_match:
    # found a match
    foo = int(regex_match.group(0))

return foo 
```

# Questions 

1. When can this expression run? On demand (when a button is pressed manually) or on every recompute?
2. How can I insert this code into a spreadsheet cell? Below isnt't working: 
    1. Right click on a cell
    2. Edit mode > try each one 
    3. Insert the code:

         ```
          #@pybegin
          import re
          doc_name = App.ActiveDocument.Label
          regex_pattern = re.compile("\\d+")
          regex_match = regex_pattern.search(doc_name)
          foo = 1
          if regex_match:
              foo = int(regex_match.group(0))
           return foo
          ```

realthunder, 2019-03-25:
> * Get the document name

Assuming you want to get the document of the current spreadsheet, simply use the following expression

```
._self.Document.Label
```

The starting dot is to refer the FC property of the current spreadsheet object. Because `Document` is not a FC property (it is a Python property), you can't directly reference it. A pseudo property `_self` is added to work around this problem. It returns the object itself, from there on (i.e. the following dots) can refer to any Python property or method.

See [here](../wiki/Expression-and-Spreadsheet#user-content-pseudo-property) for a list of other pseudo properties.

> * Use regex to extract a value from the name

One of the pseudo properties mentioned above is `._re` which exposes the Python regular expression module for just this purpose. With your code, here is the equivalent expression,

```
=._py.int(._re.search(r'''\d+''', ._self.Document.Label).group(0))
```

`.py` is another pseudo property for Python `__builtin__` modules, which is where `int()` belongs.

You have other simpler options like string split, or directly write an expression in the object's description and call `eval(obj.Label2)` to get the result as shown below.

> * Assign this number to the cell

Well, simply enter the expression in the cell correctly and it will evaluate and assign the result. For example, copy the following string to some object's description, which is an expression for an [idict](../wiki/Expression-and-Spreadsheet#idict).

```
{a=1, b=<<some text>>}
```

Then in some cell, enter the following expression to evaluate the above expression

```
eval(obj.Label2)
```

Or, retrieve one of the fields

```
eval(obj.Label2)[<<b>>]
```

> 1. When can this expression run? On demand (when a button is pressed manually) or on every recompute?

Spreadsheet has its own internal cell dependency tracking logic. Any changes in anything referred to by the cell (other cells or objects) will trigger re-evaluation of the expression in that cell, either when the spread is computed or after you finish edit the cell.

The [button edit mode](../wiki/Expression-and-Spreadsheet#button-edit-mode) expects the cell to evaluate to a function (everything is an object in Python). The function is only executed when the user press the button, not when the cell is evaluated (i.e. when the spreadsheet recomputes). To define a function, just use the normal Python syntax, `def func(blah, blah)`.

> 2. How can I insert this code into a spreadsheet cell? Below isnt't working:

For each edit mode, you must make sure the expression evaluates to the expected result type as described in the document. It is much easier to use an external editor with Python formatter to edit the code before copy to spreadsheet. See [here](../wiki/Expression-and-Spreadsheet#user-content-edit) for more details.

ceremcem, 2019-03-31:
This is the proof of concept of my use case: 

[m4x35-bolts.fcstd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3026395/m4x35-bolts.fcstd.zip)

```
##@@ A1 m4x35_bolts#Spreadsheet.cells (params)
<<Regex for parameters from doc name>>

##@@ A2 m4x35_bolts#Spreadsheet.cells (params)
<<^[mM](\\d+)x(\\d+).*$>>

##@@ B1 m4x35_bolts#Spreadsheet.cells (params)
<<Dia>>

##@@ B2 m4x35_bolts#Spreadsheet.cells (params)
._py.int(._re.search(.A2, ._self.Document.Label).group(1))

##@@ C1 m4x35_bolts#Spreadsheet.cells (params)
<<Length>>

##@@ C2 m4x35_bolts#Spreadsheet.cells (params)
._py.int(._re.search(.A2, ._self.Document.Label).group(2))
```

As a first impression, "Tree view -> Spreadsheet -> right click -> Expression actions -> Copy* / Paste" seems very handy. 

Would it be a good enhancement to add the `Alias`es to the `##@@ ... ` line too? So we could see and assign any relations in one page while we are editing the expression with our favorite editor.

ceremcem, 2019-04-02:
@realthunder Can we write a multiline Python code into the spreadsheet cell? For example, pasting the following doesn't work:

```py
##@@ D2 M3x6_bolts#Spreadsheet.cells (dimensions)
#@pybegin
a = ._re.search(.A2, ._self.Document.Label)
._py.int(a.group(1))
#@pyend
```

realthunder, 2019-04-02:
> Can we write a multiline Python code into the spreadsheet cell? For example, pasting the following doesn't work:

That is not valid Python syntax. You can have multiple line by adding `\` as the last character of a line to escape the end-of-line.

```python
a = ._re.search(.A2, ._self.Document.Label) \
   ._py.int(a.group(1))
```
Note that this line break is recognized by the expression parser, but not preserved. In other word, when you paste that line into the spreadsheet, it will merge these two lines together, similar to comments, which will be stripped. To preserve whatever you typed, enter the code as string, and use `eval(cell)` function to run the code.

realthunder, 2019-04-02:
> This logic requires a multi-line code and temporary variables.

I am not sure I understand what you mean by 'multi-line' here. You can define a function for complex logic like that, and you are free to define any variables inside the function. You can only define variable inside a function scope. If you don't intend to reuse the function elsewhere, you can define the function and immediately call it inside the same cell. 

> Would it be a good enhancement to add the `Alias`es to the `##@@ ... ` lines too? So we could see and assign any relations in one page while we are editing the expression with our favorite editor:

I'll add that in the future.

ceremcem, 2019-04-02:
### Using `[y for x,y in dict.items() if x == z]` 
Following is a valid Python syntax: 

In order to select `"some text"` from the above idict, we can use the following: 

```py
##@@ E5 M3x6_bolts#Spreadsheet.cells (dimensions)
{a=1, b=<<some text>>}

##@@ E6 M3x6_bolts#Spreadsheet.cells (dimensions)
[  y for x, y in .E5.items() if x == <<b>> ]
# => Returns ['some text']
```

However, getting the only item doesn't work: 
[image]

*BTW, how can we copy the cell errors as strings? They are not shown in report view.*

## Using functions and immediate calls
> I am not sure I understand what you mean by 'multi-line' here

I mean, the code will be complex so I can not write such a code as a one liner. 

> you can define the function and immediately call it inside the same cell.

Following action doesn't work: 

```py
##@@ E3 M3x6_bolts#Spreadsheet.cells (dimensions)
def Lookup():
    #@pybegin
    a = ._re.search(.A2, ._self.Document.Label)
    x = ._py.int(a.group(1))
    return x  

Lookup()
```

which results in: 

[image]

realthunder, 2019-04-02:
> However, getting the only item doesn't work:

It's been fixed just now.

> BTW, how can we copy the cell errors as strings? They are not shown in report view.

I'll add the console output soon.

> Using functions and immediate calls

Oops! I was wrong. You can't do that. Function calling requires name lookup and that needs a outer function scope. So you can only call function in a different cell. The function name defines the cell alias, so you can simply call it using something like `.func()`.

ceremcem, 2019-04-02:
In a regular macro, I could use the following to print a message to the Report View output:

```
import FreeCAD
FreeCAD.Console.PrintMessage("Helloo")
```

How can I use this function in expression engine? This would be very useful for debugging our own code.

ceremcem, 2019-04-02:
How can I dynamically set the spreadsheet name to retrieve a value from? For example, I have a spreadsheet named `foo`. I can get `A2` cell value of `foo` spreadsheet in `bar` spreadsheet, like so:

```
x = <<foo>>._self.get('A2')
```

How can I set the `foo` dynamically? For example, I would expect something like the following to work: 
```
table = <<foo>>
x = table._self.get('A2')
```
or 
```
table = "foo"
x = ._self.get(table + ".A2")
```

> Note: Also see #215

realthunder, 2019-04-02:
> How can I use this function in expression engine? This would be very useful for debugging our own code.

The `FreeCAD` module is aliased to `App`, so you can access it use pseudo property `_app`. BTW, `FreeCADGui` is aliased to `Gui`, which can be accessed by `_gui`.

> How can I set the `foo` dynamically?

```
table = ._self.Document.getObject(<<foo>>)
```
Assuming `foo` is the object's internal name. If you are referencing by label, and assuming there is only one object has that label,

```
table = ._self.Document.getObjectsByLabel(<<foo>>)[0]
```

Be warned though, this type of referencing has no dependency tracking, meaning that the table referenced this way may be recomputed after the current sheet. And the order may be random. To get the proper dependency tracking, use a `Link` to manage this type of dynamic reference outside of spreadsheet. To get the linked object

```
link._self.getLinkedObject()
```

If the link is broken or if it is not a `Link`, this function returns itself.

ceremcem, 2019-04-03:
Below call throws "Access denied" error, is there an important reason to restrict the `Console` module?

[image]

realthunder, 2019-04-03:
Ah, forgot about that. Not `Console` particular, but all modules are restricted by default, except those exposed by pseudo properties. You can use Python print for simple log output,

```
._py.print(<<some thing>>)
```