realthunder/FreeCAD_assembly3 #203: Auto-recover doesn't work correctly

ceremcem, 2019-03-19:
FreeCAD has crashed a few minutes ago (without a traceback, I was using in Release mode) and I re-opened my model. Recovery popup has shown and I answered as "Yes, recover my work". However, it failed to do so. Can you verify that the recovery mechanism is functioning?

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16482 (Git)
Build type: Release
Branch: LinkStage3
Hash: f670d46f2d1255867b3330052496894cf9ff7a23
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2019-03-19:
Crashed when I tried to open TechDraw page, almost the same operation with above crash. Here is the command line output: 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7ff0068b90c0]
#1  0x7ff00ac0c0b6 in Gui::MainWindow::setActiveWindow(Gui::MDIView*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x16
#2  0x7fef74c08150 in TechDrawGui::ViewProviderPage::doubleClicked() from /opt/FreeCAD/lib/TechDrawGui.so+0x20
#3  0x7ff00aa2f245 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1b5
#4  0x7ff003156a00 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xe10
#5  0x7ff00351ac8e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e
#6  0x7ff00363fba3 in QAbstractItemView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e3
#7  0x7ff00368146c in QTreeView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3c
#8  0x7ff0026390a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#9  0x7ff0030ff52c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#10  0x7ff003107ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#11  0x7ff00a8b8821 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#12  0x7ff002638f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#13  0x7ff003105ccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#14  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7ff0031811a9]
#15  0x7ff00317fb5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#16  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7ff0031a9502]
#17  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7feff9bce7f7]
#18  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7feff9bcea60]
#19  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7feff9bceb0c]
#20  0x7ff002669854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#21  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7ff0031a95d6]
#22  0x7ff0026377ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#23  0x7ff002637b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#24  0x7ff00263dbd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#25  0x7ff00a83467a in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x18da
#26  freecad-git(main+0x862) [0x55602867b7b2]
#27  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7ff000ef32e1]
#28  freecad-git(_start+0x2a) [0x55602867ca0a]
```

Note: If you need the model, I may send you privately.

ceremcem, 2019-03-19:
Okay, I found it! Here is the reproduction steps: 

1. Create a new document 
2. Add a TechDraw page 
3. Click the TechDraw page, press space (like you want to make it invisible)
4. Double-click the TechDraw page
5. FreeCAD crashes. 

# Workaround 
Double-check that the techdraw page is not invisible before you click it.

realthunder, 2019-03-28:
The TechDraw bug is fixed. And the auto recovery is working. Your problem may be caused by crash in the middle of recovery saving, as the source code seem to show that property files are saved using separate threads. I guess this can be made safer by saving to a temp file and rename back in the main thread. I'll see what I can do.

realthunder, 2019-03-29:
I made two commits about the auto recovery feature. One adds support of opening linked document during recovery, the other fixes and improves the no-compress auto saving mode. BTW, only in this mode, FC will use multi-threading while saving property files, and I don't think you enabled this no-compress mode (using parameter editor, BaseApp/Preferences/Document/AutoSaveCompressed, set to false). 

This means that the problem you have is probably not what I said before, that is, crash in the middle of saving. In the future, once you have this problem, you can go to your `/tmp' directory and find the second latest transient directory with name starting with `FreeCAD_`, and give me the files inside.

ceremcem, 2019-03-29:
I can't reproduce the crash even when I discard [the fix](https://github.com/realthunder/FreeCAD/commit/cdecc9bb1b2bbe5e3d57540663b2ddf1d7608d4d) and rebuild, so this means that both you are correct (it's not related with techdraw page visibility) and I'm not able to test the autorecover functionality. 

Next time I'll hand over the second latest transient directory (by modification time, I guess) contents.

closed 2019-03-29 by ceremcem